nodemcu/nodemcu-firmware #440: "not enough memory" to compile bmp085 module using 096

8n1, 2015-05-24:
It's not possible to compile the bmp085.lua using the latest build(0.9.6 - 20150406) from the dev096 branch because of "not enough memory". 
It works just fine under 0.9.5(20150213).

marcelstoer, 2015-05-24:
You can [build the NodeMCU firmware yourself](https://github.com/nodemcu/nodemcu-firmware#build-option) and disable all the modules you don't need thereby freeing some space. Or you can build the firmware with http://frightanic.com/nodemcu-custom-build/. Disclaimer: that's my site.

8n1, 2015-05-25:
Hey nice one! I already thought about building the firmwary myself. Thanks for sharing. I'll give it a try.

znsoft, 2015-05-27:
try to split bmp module to many small linear modules less then 50 lines of code, and remove comments.

Offtop: WOW i`ll get to favorite this one web builder.  Cool !

8n1, 2015-05-28:
hmm. havn't thought about splitting the module itself. Anyhow this shouldn't be neccessary to use it, right?
I already noticed that splitting up code into smaller parts helps a lot. And in fact it's the only way I' m able to use a large number of modules at once right now on 0.9.5. (actually not at once, but one by one)

The other things you mentioned are not new to me too, but thanks!

8n1, 2015-08-03:
This is no longer a issue since the the latest firmware builds have a lot more ram available. 
Which is pretty awesome, btw.

closed 2015-08-03 by 8n1