nodemcu/nodemcu-firmware #441: Use of Station AP

jtfell, 2015-05-25:
Setting 2 nodes in STATIONAP mode then having them both set up an AP (with a UDP server) then connect to each other's AP and send to the other UDP server isn't working for me. Am I correct in thinking this is possible? If I am, what am I doing wrong?

My code is below, and I switch the ssids around on the other node so that isn't the issue.

    scanning = false
    nodeID = node.chipid()

    -- Create AP
    cfg={ssid = "csse4011" .. nodeID, pwd = "csse4011"}
    wifi.setmode(wifi.STATIONAP)
    wifi.ap.config(cfg)
    -- Start UDP server
    s=net.createServer(net.UDP) 
    s:on("receive",function(s,c) print(c) end)
    s:listen(4011)

    wifi.sta.config("BaseNode_csse4011", "csse4011")
    wifi.sta.connect()
    tmr.delay(100000)

    print(wifi.sta.getip())
    print(wifi.sta.status())

    -- Create UDP connection
    cu=net.createConnection(net.UDP)
    cu:connect(4011,"192.168.4.1")
    cu:send("hey")

    tmr.delay(100000)
    print(wifi.sta.status())
    wifi.sta.disconnect()
    print(wifi.sta.status())

dperlman, 2015-05-25:
You can't use tmr.delay like that. You have to properly use event-driven networking. Look up a tutorial on node.js online, that is where the networking model is derived from.

In general, if you are using tmr.delay at all in nodeMCU code you are probably doing something wrong and it won't work... There has been some discussion on the message boards about the need for a basic introduction to the right way to think about event-driven network coding but such a guide doesn't exist yet. In the meanwhile, the main thing to keep in mind is that if you open a network connection and you want something to happen after the connection opens, or after the data is sent, or after the response is received, or whatever, the way you do that is by putting code in the callbacks for those events. If you want something to happen a certain amount of time after something else, you use a tmr.alarm() for that. Those two rules should cover most of your scenarios, so stick with that and don't ever use tmr.delay()

jtfell, 2015-05-25:
Yeah I was using alarms originally, but I was trying to simplify the code I was posting here. 

This is closer to what I'm trying to do:

```lua

scanning = false
nodeID = node.chipid()

-- Create AP
cfg={ssid = "csse4011" .. nodeID, pwd = "csse4011"}
wifi.setmode(wifi.STATIONAP)
wifi.ap.config(cfg)
-- Start UDP server
serv=net.createServer(net.UDP) 
serv:on("receive",function(s,c) print(s,c) end)
serv:listen(4011)

-- Setup timer callbacks
tmr.alarm(0, 500, 1, function()
   if not scanning then
       scanning = true
       blink()

		-- Connect to base server
		wifi.sta.config("BaseNode_csse4011", "csse4011")
		wifi.sta.connect()

		print(wifi.sta.getip())
		print(wifi.sta.status())

		-- Create UDP connection
		conn=net.createConnection(net.UDP)
		conn:connect(4011,"192.168.4.1")
		conn:send("hey")

		tmr.delay(100000)
		print(wifi.sta.status())
		wifi.sta.disconnect()
		print(wifi.sta.status())
   end
end)
```

TerryE, 2015-05-25:
@jtfell , this isn't an issue with nodeMCU per se, more that you need advice in how to code your applications. That's what forums like esp8266.com aim to provide. This list is for errors in the firmware. Please close this issue and use a forum. Terry

closed 2015-05-27 by jtfell