nodemcu/nodemcu-firmware #442: Check restart reason

vudangngoc, 2015-05-25:
I think it's good to know why NodeMCU restart, this information will help developer handle behavior of NodeMCU more easily, avoid causes make NodeMCU restart and improve reliable of ESP.

dperlman, 2015-05-25:
My understanding is that nodeMCU restarts because the watchdog timer times out. This means either that user code entered a blocking loop which didn't allow the timer to reset; or else the firmware was rendered non-functional by memory corruption resulting from heap overflow. In either case, there's no practical way for the firmware to track the cause of its own failure. However, if you watch the serial console output, there is almost always an error message before the restart and you can figure out what happened from that.

vudangngoc, 2015-05-26:
Can we determine restart cause is power cutting or watchdog time out?

cal101, 2015-05-26:
Moin,

watchdog resets will print "wdt reset".

Cal

drv53b6, 2015-06-05:
node.restart(restart_code)
print(node.rcode())
or
print(rcode) rcode=nil

TerryE, 2015-06-05:
@drv53b6 -- except that `node.restart()` doesn't take any parameters and the **node** library doesn't contain a  `node.rcode()` method.

drv53b6, 2015-06-05:
@TerryE must do it, it will be very useful

vudangngoc, 2015-06-26:
@cal101 : where it print "wdt reset"? any news about this issue?

jmattsson, 2015-06-26:
There is also a ROM function called rtc_get_reset_reason() which returns the reset reason. 4 is the code for wdt reset, 3 for reset via the reset pin. I can't remember what the code for a cold boot is off the top of my head.

vudangngoc, 2015-06-26:
@jmattsson : could you please give me a piece of example code?

MarsTechHAN, 2015-06-26:
@vudangngoc When ESP8266 restart it will print information at 74880bps

Air000, 2015-07-12:
@jmattsson where to find the definitions of rtc_get_reset_reason()'s return value?

jorgempy, 2015-08-05:
this should be closed, as #512 has been merged. also it should be documented somewhere.

closed 2015-08-06 by vudangngoc