nodemcu/nodemcu-firmware #444: File system isn't usable on dev096

pastukhov, 2015-05-26:
```lua
>     file.open("init.lua", "a+")
>     file.writeline('foo bar')
stdin:1: open a file first
```

creationix, 2015-05-26:
Do you have one of the new nodemcu devboards?  They require special flashing tools in windows to get a working filesystem.

pastukhov, 2015-05-26:
No. I have esp-01, esp-12, esp-201 and nodemcu devboard v0.9. And i am on ubuntu.

jonn26, 2015-05-26:
+1 I am seeing this as well on esp01 & esp-12.

drv53b6, 2015-05-27:
file.format()   -- ?

dnc40085, 2015-05-27:
+1 confirmed on esp-01 esp-07, compiled on lubuntu with esp-open-sdk.

I cloned the current dev096 branch, compiled, and burned, then ran file.format() then used file.fsinfo() and it returned `stdin:1: file system error` on both the esp-01 and the esp-07

pastukhov, 2015-05-27:
file.format() doesn't help
The problem  is in 5261fb194853f7f56c9578fa582f9984c3cd955e
Without it everything is fine.

pastukhov, 2015-05-27:
file.format() log with debug:
```
> file.format()
_flash_used_end:40268eb6
fs.start:4026c000,max:10000
CACHE_ALLO: allocated cache page 0
page index byte len:         512
object lookup pages:         1
page pages per block:        128
page header length:          5
object header index entries: 234
object index entries:        253
available file descriptors:  3
free blocks:                 1
mount res: 0
erase 4026c000:00001000
erase 4026d000:00001000
erase 4026e000:00001000
erase 4026f000:00001000
erase 40270000:00001000
erase 40271000:00001000
erase 40272000:00001000
erase 40273000:00001000
erase 40274000:00001000
erase 40275000:00001000
erase 40276000:00001000
erase 40277000:00001000
erase 40278000:00001000
erase 40279000:00001000
erase 4027a000:00001000
erase 4027b000:00001000
CACHE_GET: have cache page 0 for 0000
CACHE_GET: have cache page 0 for 0000
_flash_used_end:40268eb6
fs.start:4026c000,max:10000
CACHE_ALLO: allocated cache page 0
page index byte len:         512
object lookup pages:         1
page pages per block:        128
page header length:          5
object header index entries: 234
object index entries:        253
available file descriptors:  3
free blocks:                 1
mount res: 0
format done.
```
file.fsinfo()
```
file.fsinfo()
total: -63882, used:0
stdin:1: file system error
```

creationix, 2015-05-28:
I can also confirm.  I just tested on the ESP-201 boards I got in the mail today.

dnc40085, 2015-05-29:
Seems like the last commit to work was 8f925e6

dnc40085, 2015-05-29:
Just compiled and tested dev096 and the filesystem now works.

closed 2015-06-22 by pastukhov