nodemcu/nodemcu-firmware #446: uart.on put device in endless loop

ntfs1984, 2015-05-27:
Reading from UART may put device into loop without end. Then all commands doesn't work except of "node.restart()"

Example of code:

uart.on("data", "\n", 
      function(data)
        print("receive from uart:", data)
        if data=="swon\r\n" then 
          uart.write(0,"Received ON command");
        end        
end, 0)

jonn26, 2015-05-27:
Try adding "if data ~= nil then" line like below:

uart.on("data", "\n", 
function(data)
if data ~= nil then
print("receive from uart:", data)
if data=="swon\r\n" then 
uart.write(0,"Received ON command")
end
end

end, 0)

ntfs1984, 2015-05-28:
Unfortunately, doesn't work.
It stops even if

uart.on("data", "\n", 
function(data)
;
end, 0)

- - - 
I tried to replace "\n" to other symbols - still doesn't work. Checked with two modules.

MarsTechHAN, 2015-06-19:
= =Dont forget ...When uart.on(.....,0)<- This set to zero, IT MEANS THE DATA WONT GO THROUGH LUA

As for why node.restart() will work= = I believe that's a bug

MarsTechHAN, 2015-06-19:
<Reading from UART may put device into loop without end. Then all commands doesn't work except of "node.restart()">

We have tested, even node.restart() doesn't work

TerryE, 2015-06-19:
+1.  Whilst an `uart.on("data"..,0)` is in scope, you've effectively disabled the `lua -i` mode.

ntfs1984, 2015-06-19:
I believe, this is a bug, because on a fact, simple lua firmware might block device for further work.
Of course we can use a trick - two-second delay directly since power on or something like that.

Mic-Key, 2015-06-19:
@ntfs1984 look at the [example](https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#uarton), especially at:
```
if data=="quit\r" then
    uart.on("data") 
end 
```
M.

MarsTechHAN, 2015-06-21:
you can turn the scripter on... That's fine

uart.on(.....,1)<- This set to one

And the data will also go through scripter

TerryE, 2015-12-12:
This is a feature not a bug.

closed 2015-12-12 by TerryE