nodemcu/nodemcu-firmware #448: lm35 lua script

AhmedDjebali, 2015-05-28:
Hello guys,

I have lm35 temperature sensor, can I change the ds18b20 script to use it with the lm35 sensor? if yes, what are these changes? or is there an lm35 lua script?

Thanks previously 

--ds18b20 script
 require('ds18b20')
 -- ESP-01 GPIO Mapping
 gpio0 =3
 gpio2 =4
 ds18b20.setup(gpio2)
 t=ds18b20.read()
 print("Temp:" .. ds18b20.read() .. " C\n")
 if(t==nil) then
 t=0
 end
 tmr.alarm(0,30000, 1, function()
 t=ds18b20.read()
 conn=net.createConnection(net.TCP, 0)
 conn:on("receive", function(conn, payload) print(payload) end )
 conn:connect(8080,"DOMOTICZ IP")
 conn:send("GET /json.htm?type=command&param=udevice&idx=32&nvalue=0&svalue=" .. t .. " HTTP/1.1\r\nHost: www.local.lan\r\n"
 .."Connection: keep-alive\r\nAccept: */*\r\n\r\n")
 end)

dnc40085, 2015-05-29:
Hello, 
   The NodeMCU issues section is for reporting bugs in the firmware.
You would probably get a faster response by starting a thread on [esp8266.com](http://www.esp8266.com) in the [Random Lua Talk](http://www.esp8266.com/viewforum.php?f=24) forum, as there are more people in the forums than here in the issues section.

MarsTechHAN, 2015-06-19:
DS18B20 is digital sensor, but LM35 is analog sensor. The program is totoally different...So...We will have that lib inrecently

nickandrew, 2016-02-10:
Not a bug, so closing.

closed 2016-02-10 by nickandrew