nodemcu/nodemcu-firmware #449: Creating a nodeMCU Lua (unofficial) FAQ

TerryE, 2015-05-28:
_Forgive this post as an issue but I wanted feedback and comment from this audience with only has part overlap with the [equivalent topic on esp8266.com](http://www.esp8266.com/viewtopic.php?f=24&t=3311&p=18770)._

This is really by way of an RFC.  I've started drafting an unofficial FAQ on a local MediaWiki instance.  It's only about 30% complete and I am still in rough brain-dump mode.  It still needs proper proofing and polishing, but I've dumped a copy in my github nodemcu-firmware fork: [Unofficial FAQ](url=https://github.com/TerryE/nodemcu-firmware/wiki/Unofficial-FAQ) for those that might want to take a skim.   I don't think that an ultimate master should belong here so at some point I will can it , if and when we transfer the then content.

My inspiration for this is the incredibly useful [Lua Unofficial FAQ \(uFAQ\)](http://www.luafaq.org/).  What I want to put in here are Q&A for all of the questions that new to nodeMCU Lua programmers reqularly trip over and ask on this site and on the nodMCU wiki issues list.

*    Please use the esp8266.com topic reference above is you have an account there  to discuss / correct the content / suggest extra additions.  Even a list of placeholder questions would be good.
*    Alternative post here as long as the nodeMCU guys are comfortable with this.
*    Does anyone have a better suggestion (i) where we should do initial work, and (ii) where the list should ultimately reside? Should something like this be part of the standard node MCU documentation?

As you will see I am still on a voyage of discovery on how this Lua RTS works, but there's detail in here that I just haven't found anywhere else and have only worked out through re-engineering the source.

If anyone wants to chat / skype / email on specific points then drop me a mail.  My mail domain is as per my website on my profile less the "blog." and my mailbox is just Terry at this.

TerryE, 2015-06-02:
I've relocated this extended and updated page to the esp8266 site and changed the first post reference.  I won't post to this issue again unless others feel that some version of this content should be pulled into the formal nodeMCU documentation, and if I get no other comments, I'll close this issue in a week or so.

creationix, 2015-06-02:
Thanks for making this.  I see lots of useful stuff in there.

marcelstoer, 2015-06-04:
When the topic of building your own firmware is discussed (e.g. around ["However, if you choose to use a custom build..."](http://www.esp8266.com/wiki/doku.php?id=nodemcu-unofficial-faq#how_is_coding_for_the_esp8266_different_to_standard_lua)) it'd be helpful give some guidance as to which NodeMCU branch to base the build on.
I think I understand what `master` and `dev096` are used for and I think I get what `dev` was _intended_ for. Yet, I'm surprised new PRs are merged right into `dev096`. So, I obviously do _not_ get it ;-)

cal101, 2015-06-04:
Moin Marcel,

if you have a good idea about the intended usage please tell me. Seriously. 
Sometimes thought I understand it but then ...

Cal

TerryE, 2015-06-04:
@marcelstoer,  Marcel, I think that *somewhere* we could really do with a plain instruction on how to "roll-your-own" build, probably on 4 variant platfroms:

*   On a standard Linux platform
*   On WinX
*   On iOS
*   In the cloud (i.e. your service) 

and no I have no clue how vowstar and the other nodeMCU guys manager their tags.  I would have though we'd have **master**, a **release095** (and any other formal releases)  and maybe one or two dev forks, but that's not how it operates, is it?

cal101, 2015-06-05:
Moin,

Thinking more about the branches I think its a major issue.
If we don't know what the branches are good for how should the casual user know.

@marcelstoer wrote a great interface for dummy users to build their custom firmware but what should they use?
@marcelstoer did work to get more detailed build information and version information into the firmware
People need to be told that this information or some other version identification is essential 
for bug reports. If not already that version string must be accessible via "node".
This kind of build information should be added to the release builds as well.

@vowstar 
@dnc40085
@nodemcu 
@devsaurus 
@MarsTechHAN 
Can anybody from the core team or from the major contributors please enlight us about the intended
purpose of the nodemcu branches and the idea when providing new releases?

People are asking us what they should use and we want to be able to give better advice.

Cal

devsaurus, 2015-06-05:
There were #176 by @nodemcu and #256 by @GeorgeHahn. Thus I usually PR against `dev` branch until further notice.
This also let's me assume that `dev` should be regarded as testing while `master` is more like production.

marcelstoer, 2015-06-05:
> @marcelstoer did work to get more detailed build information and version information into the firmware

Rather than fiddling with https://github.com/nodemcu/nodemcu-firmware/blob/master/app/lua/lua.c#L140 I just modify `NODE_VERSION` and `BUILD_DATE` in `user_version.h` to be something like 

```
#define NODE_VERSION "NodeMCU custom build by frightanic.com\n\tbranch: master\n\tcommit: 0ad574705d1a22b9ebc62bc198e6db9e792244e0\n\tSSL: false\n\tmodules: node,file,gpio,wifi,net,pwm,i2c,tmr,adc,uart,ow,ws2812,cjson\n"
#define BUILD_DATE "\tbuilt on: 2015-06-05 11:56\n"
```
which gets resolved as 

```
NodeMCU custom build by frightanic.com
   branch: master
   commit: 0ad574705d1a22b9ebc62bc198e6db9e792244e0
   SSL: false
   modules: node,file,gpio,wifi,net,pwm,i2c,tmr,adc,uart,ow,ws2812,cjson
   built on: 2015-06-05 11:56
  powered by Lua 5.1.4
```
when the firmware is booted.

I realize that this isn't elegant but I wasn't ready to make a PR which would allow for more `user_version.h` variables (e.g. `BRANCH`, `COMMIT`) to be used in `lua.c:print_version`. The modules and SSL information is actually already there and would simply have to be pulled from their respective sources (`user_modules.h`, `user_config.h`). But as I said I wasn't ready to modify NodeMCU C code - partly because I didn't know into which branch the changes should be merged.

dnc40085, 2015-06-05:
@cal I don't know for sure... but before I submitted my first PR I think I read somewhere in the "dev096" branch that the NodeMCU dev team was following the SDK's version numbering scheme, hence NodeMCU ver 0.9.5 and with the jump to SDK version 1.0 the NodeMCU team decided that it would split it's version scheme from the SDK and that the next version of NodeMCU would be ver 0.9.6.

Other observations:
* the last release (04062015) said in the comments "bump to 0.9.6" and that release was compiled from the dev096 branch
* I've seen PRs from dev -> dev096, but not dev096 -> dev.

All this leads me to believe that the dev096 branch is the new version that's in development. but... there's this PR(#423) that was submitted to the master branch and it has a comment by vowstar that makes me think otherwise:
> dev096 is a short-life version, please PR to dev and we will merge it. 

To tell the truth, the only reason all my PRs were submitted to the dev096 branch is because 096>095

marcelstoer, 2015-06-05:
@devsaurus Exactly! `CONTRIBUTING.md` is quite clear

> Pull requests for new features and major fixes should be opened against the dev branch.

That's why I was so confused when #465 was merged into dev096.

cal101, 2015-06-05:
Hmm, I checked. "dev" has the same bug. WTF Help BEEP beep ....

cal101, 2015-06-08:
Is it worth to put so much effort in nodemcu if the core team isn't willing to discuss some basic stuff?
Is it just me asking and seeing a need?

vowstar, 2015-06-08:
I'm sorry for that, we are developing v1.1.1 branch and waiting new esp8268 ic. Recently we have little spare time, still we are so small, but we will try our best on nodemcu.

TerryE, 2015-06-08:
> ... waiting new esp8268 [integrated circuit] ...

Interesting.  It looks as if my 12 months estimate was wrong.  I realise that you guys are almost certainly constrained by a Non-Disclosure Agreement (NDA).  However, if there is a new ESP8266 chip about to be released  (with 128KB+ RAM say) in the next 6 months, then maybe our (community) development priorities would change as well.

vowstar, 2015-06-08:
I will contribute the wiki, but my English is poor, if I make mistake, please help me :

TerryE, 2015-06-08:
@vowstar, your English is a _lot_ better than my Chinese :smile:! We are interested in your knowledge and expertise and any contributions will be very welcome!  One of the more fluent English speakers can always tidy up the content if needed.

vowstar, 2015-06-08:
Yes, new IC is comming, almost 200kB RAM. May support WiFi+BLE.

TerryE, 2015-06-08:
Wooooo !!

alonewolfx2, 2015-06-08:
@vowstar soon? 1 month ,2 month?

TerryE, 2015-06-08:
I suspect that Espressif are shipping evaluation samples to key developers and integrators, so if all goes will it should be less than a few months before they start shipping in volume.  We saw the same thing happen with Upton's announcement of the RPi2.

As I said the nodeMCU team have probably signed an NDA, and we should push them too hard.  It's really up to Espressif to make any official announcement,

MarsTechHAN, 2015-06-11:
New chip still will take some time... Information from Espressif, they said the chip mostly like to release at the end of this year.

TerryE, 2015-06-11:
@MarsTechHAN, Martin thank-you for this clarification.  I feel that Espressif have produced an excellent product in the ESP8266, but they need to have a roadmap for maintaining that competitiveness, and as do supporting of suppliers such as nodeMCU.  This timescale doesn't surprise me, but at my age six months passes all too rapidly! :laughing:

alonewolfx2, 2015-06-11:
@TerryE you are right :D main question is new chips will come with the new SDK or current and developing sdk.

vowstar, 2015-06-21:
I will take photos of espressif's roadmap and upload here.

TerryE, 2015-06-21:
@vowstar, thanks and I will move them into the esp8266.com Wiki.

vowstar, 2015-06-22:
[image]

TerryE, 2015-06-22:
Yup, but at the moment @marcelstoer Marcel is providing his cloud service as a favour.  It is rapidly coming the standard unofficial comminity tool but if so the it would better open sourced, in my view.

vowstar, 2015-06-22:
@TerryE Very nice cloud service.

zerog2k, 2015-07-26:
@vowstar
@dnc40085
@nodemcu
@devsaurus
@MarsTechHAN

Back to the questions about branches and version, which don't appear to have been completely answered/resolved.

I see that there are dev branches that correlate to sdk version, e.g., dev096 -> esp-sdk 0.9.6, but this brings to mind a few thoughts:
1. espressif is long past 0.9.6, and at 1.2.0. Why is there any development on this branch at all? Should this all be on dev120 now (and dev096 frozen)?
2. I'm not sure I see why the nodemcu version tracks the esp sdk version. Example: I built dev120 and master branch just fine using esp sdk 1.2.0. If there is a reason for this, it is not stated clearly.

I think alot of confusion can be eliminated simply and quickly with more details in CONTRIBUTING.md. Info needed would be:
1. what branches are used for and rationale
2. where fixes go vs new features/modules
3. any other style, testing, or conventions

Thanks for your time on this. There are many folks here who want to help polish the rough edges on nodemcu and make it awsome, and most important of all for any IoT platform, *STABLE*.

vowstar, 2015-07-26:
@zerog2k 
1. Dev120 closed many source code and opensource lwip can't working well yet. Dev120 is test branch and not stable.
2. Master branch based on espressif's SDK 0.9.6b1, no matter what SDK you used, if you don't changed Makefile, it always use the lib from SDK0.9.6b1.

All pull request better pull to dev branch for experimental, and master branch is stable than dev branch(not stable yet). Dev096, dev120 is temporary branch, under developing. All new feature first pull to dev branch, and then pull dev to both dev096 and dev120 for testing. When it passed testing, it will merge into master.

nodemcu, 2015-07-26:
Hi, there
thank you very much, for everything!
about the branch and version:
previously what I assume the branches for:
master: stable.
dev: new feature and module. after some test, some feature will go to master.
dev096: was a branch to test a "NEW" release of sdk 0.96 at early age...
and when sdk > 1.0.0 released, it eat a lot memory, so we decided not to upgrade to it.
after the great patch from @jmattsson, I think it's time to upgrade partially to new sdk, except the closed-source part.

but obviously these branches setup caused a lot problems and misunderstanding. 
(this is my first opensource-github-serious project. don't know how to manage a project in a good manner, sorry for that.) 

so I am here call for some help:
suggestions of how to organize this repo, branches setup?
and how to sort these branches out and make it organized and clean and easy to understand?

eyaleb, 2015-07-27:
I will now dump my brain here...

My approach would be to keep things simple. As such, a 'master' and 'dev' is mostly enough. Development goes into 'dev' and when ready it goes to 'master'.

It is useful to also have a 'staging' branch which has very limited check-in permissions, and which is used to validate that all is well before pulling into master. It pulls stuff from 'dev', and if a problem needs fixing here it is later pushed to 'dev'.

This way 'dev' can continue to receive patches while 'staging' is prepared for release. 'staging' exists only for short periods from when a decision to make a new release is  taken until the release is made.

Release candidates should be made from 'staging'. Final release from 'master'.

'master' should have very limited check-in permissions (and only pulls from 'staging') and should always be release quality.

I see 'dev120' as a temporary branch, same as everyone else keeping a private branch before requesting a pull. It has no stability and no promises. This will be gone once that specific development is completed (same as 'dev096' should have been gone long ago) and a final pull is made. Then 'dev' is where life continues.

I expect more such temporary branches, where development is made on large projects that is difficult to do in 'dev' without much breakage and conflict.

BTW, I am aware of (and used) a different approach, where there is no 'staging'. Instead, when 'master'  is good for a release, a branch is created and named for that release (e.g. '20150727') and 'master' continues to be updated. 'master' is effectively a staging, mostly stable branch. The release branch will later spawn dot-release branches (for required post-release critical fixes).

Enough for now...

TerryE, 2015-10-23:
This thread wondered off into a discussion of dev versions and bringing up the firmware to a current version, which has now largely been addressed elsewhere.  

I want to bring this discussion back to the FAQ itself.  I wrote this months ago.  Things have moved on (e.g. Johny's patch) and my understanding of the Lua firmware improved.  So IMO, it's now time for a major restructuring of the FAQ.  

*    I want to add a lot more content, but it's already too unwieldy as a single document, so I want to split it into a core document and more specialist sections.  Beginners seem to be unwilling to go through this but still litter esp8266.com and here with issues which they trip over. so I want to try to separate out the more complex topics and improve the readability of the core stuff.
*    This whole issue of the firmware and the SDK being generally asynchronous / event driven needs to be split offinto its own tutorial with better examples.
*    I need to have a section on building and customising the firmware.  I have already added a section [How to setup a VM to host your toolchain](http://www.esp8266.com/wiki/doku.php?id=toolchain#how_to_setup_a_vm_to_host_your_toolchain) to the **toolchain** page.  But we could really do with a tutorial on how to customise and build and flash the firmware.
*    Some explanation of memory allocation and usage would probably help.
*    Persistence and logging techniques ditto.

@jmattsson @marcelstoer @Aeprox @eyelab any more specific suggestions?  I currently do my dowiki stuff on a local instance before uploading to esp8266.com, but I will set up a publicly accessible drafting instance on my `ellisons.org.uk` service so that I can do this work and get feedback before updating to public master.

marcelstoer, 2015-10-23:
Was just reading the FAQ when I got your notification ;-) I'll be releasing a Docker image in the next days which eases building the firmware on your own machine (i.e. without setting up full fletched Ubuntu VM).

Aeprox, 2015-10-23:
I was going to do a blogpost about the config/build/flash process this weekend anyway, so just point me to a place to upload it and I'll do a rewrite.
One other thing that might be useful is an overview of memory usage of the configurables. eg. how disabling SSL adds 40KB of free flash. Nearly gave up on 1.4.0 until someone mentioned that.

Fully agree on the rebase! Maybe start a different issues asking for bugs/feedback before merging dev to master though? I noticed the adc.readvdd "bug" when updating my project but I'm sure there's more out there.

@TerryE How about moving to ghpages? Has been discussed before but that would give us a) full access, b) platform to discuss changes to docs, right? Imho this lowers the effort of quickly updating the docs after a question/issues points out something's missing in there.

TerryE, 2015-10-23:
@Aeprox Michael, I want this info to be in a central repository that we all have access to. GH markdown is different to MW (as in WP) markdown and again different from Dokuwiki markdown.  I don't mind the one-off pain of migrating from one to another, but not keeping two variants in sync. I an a contributor to nodeMCU but not an admin so whilst I have write access to its wiki, I can't give you access and the guys with admin access have "gone walkabouts".  

The nice thing about the GH wiki is that its just a GH repo, with all of the GH features, but that the wiki engine understands a set of flavours of MD including Mediawiki MD.  

What we need is a decent body of documentation including examples, and an active set of contributors working on this, rather than one bod.  

IMO, the main problem with this repository is that its continually being spammed by beginners asking novice Qs.  (I don't have a problem with beginners doing this, but just not here.  It means that we can't see the wood from the trees).   What I would love to have is a repo where any write access (including issues) is limited to active contributors.  

Ultimately I would prefer to see this as part of the official documentation as part of the distro, but I do thing that the collaboration vehicle for developing this content is a different issue.  As is the development format (e.g we might develop in a MD/MU such as GH MD or MediaWiki MU, but ultimately publish in html; the former would be in change control).

Aeprox, 2015-10-24:
I'm on-board for helping in any way I can, but I'll just follow the lead. In all honesty I've just graduated and I'm just tagging along, learning a lot and watch you guys in awe. I have very little experience to draw from when it comes to managing community projects.

I'd get started fixing the GH wiki straight away if it wasn't such a mess right now. And to clean that up, we'd need to clean up the branches or it won't be clear what applies to what version (re-basing to 140 would certainly do this). To do that we'd need help from the chinese guys..

The call for new administrators has been a while ago, but we haven't heard a lot from them since then. I'd start a community fork if I weren't 100% inexperienced in doing so, because their current lack of communication/hand-over is seriously stalling the project imho. Don't get me wrong, they did an awesome job creating the project! Life goes on and priorities change.

jmattsson, 2015-10-25:
For the API docs I'd like to see that split into multiple pages, and just keep the module index on the main page. That alone would make the docs seem less intimidating I think, both from a user and maintainer point of view.

I'd also prefer to get rid of the dev096 page and the concept thereof, and simply add an "Available since:" field to each module/function as applicable. That way there'd be no need to copy/paste things around when there are realeases, and the docs might actually stay relevant (which is always a challenge, no matter what project).

With the dev branch now being on a modern post-1.0 SDK, I'd be in favour of tagging an actual release and consigning 0.9.5 to pleasant memories alone. The project needs momentum as much as general tidying up, and being bold about a new release is probably not a bad choice.

marcelstoer, 2015-10-25:
> add an "Available since:" field to each module/function

That would help a bit but is IMO too cumbersome.

For the reasons stated in https://github.com/nodemcu/nodemcu-firmware/issues/577#issuecomment-125326740 I still prefer to have the documentation in the repository rather than in a wiki.

TerryE, 2015-10-25:
> add an "Available since:" field to each module/function

This is what PHP does and it works well.  I have no problems with using a MD or MU flavour for the root documentation, but the end-developers who want to use Lua want access to a rendered version.  

BTW most wikis are repositories under the hood.  For example the github wikis just use git.    Yes also think that we need decent API documentation.  Even so, I don't entirely agree that SO and esp8266-style forums should be the *only* alternative.  I wrote the Unofficial FAQ because I got bored answering the same Qs over and over again.  That's the sweet spot that FAQ hit.

TerryE, 2016-01-14:
This issue has got a load of off-issue posts that are now all historical, so I am going to close it and start a clean slate

closed 2016-01-14 by TerryE