nodemcu/nodemcu-firmware #451: SPI MODE3 doesn't seem to be working

goseese, 2015-05-29:
I am using NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4

Is SPI Mode # working? CPOL = 1, CPHA = 1? 
spi.setup(1, spi.MASTER, spi.CPOL_HIGH, spi.CPHA_HIGH, spi.DATABITS_8, 0)

According to the docs it looks like it's supported
https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#spisetup

However, when I view it with the salea logic analyzer the SCK starts out low, not high.

dnc40085, 2015-05-29:
Hello, 
   I believe this feature has not yet been implemented.

From the file app/driver/spi.c:
```
void spi_master_init(uint8 spi_no, unsigned cpol, unsigned cpha, unsigned databits, uint32_t clock)
{
...
    //set clock polarity
    // TODO: This doesn't work
    //if (cpol == 1) {
    //    SET_PERI_REG_MASK(SPI_CTRL2(spi_no), (SPI_CK_OUT_HIGH_MODE<<SPI_CK_OUT_HIGH_MODE_S));
    //} else {
    //    SET_PERI_REG_MASK(SPI_CTRL2(spi_no), (SPI_CK_OUT_LOW_MODE<<SPI_CK_OUT_LOW_MODE_S));
    //}
    //os_printf("SPI_CTRL2 is %08x\n",READ_PERI_REG(SPI_CTRL2(spi_no)));
...
}
```

drv53b6, 2015-05-30:
this one:
https://github.com/cpldcpu/light_ws2812/blob/master/light_apa102_AVR/Light_apa102/light_apa102.c
worck ok

MarsTechHAN, 2015-06-20:
There is no change in SPI drive recently...

But we have received report that u8g spi no longer work in new version...So we will use equiment to analysis the communication tomorrow... And wish we can reach a conclusion...

Duality4Y, 2015-08-25:
What is the Status on this? just wondering :)

devsaurus, 2015-12-13:
SPI module needs some rework regarding MODE3.

devsaurus, 2016-01-14:
Thanks a lot @jfollas!

closed 2016-01-14 by devsaurus