nodemcu/nodemcu-firmware #452: node.getregistry() to mirror debug.getregistry()

TerryE, 2015-05-29:
The lack of a debug implementation within the NodeMCU firmware is a bit of a limitation.  The one function that I really miss is `debug.getregistry()` as this pretty essential in understanding where memory leaks are occurring.  A simple work around would be to move this function into the node library.

With this, its quite easy to use a table walker to analyse what it being squirrelled away in the Lua registry and consuming RAM. 

I'll implement this on my own build in the next week or so and if it proves useful, then I'll issue a pull request,

drv53b6, 2015-05-30:
```
for k,v in pairs(_G) do
    print("Global key", k, "value", v)
end
```

TerryE, 2015-05-30:
@drv53b6, why is your example relevant to this issue?  The registry is not the same as _G.  It's not visible by default so you have no idea what library modules are dumping into it.  Try doing this on your PC with `lua -i` and use `debug.getregistry()` instead of `_G` in your for loop to see what I mean.

drv53b6, 2015-05-30:
i dont have debug build borned

TerryE, 2015-05-30:
Download a standard Lua build for your PC and use it to test compile your code etc.  My point is that even on a PC install of Lua, these are different.

drv53b6, 2015-05-30:
@TerryE 
only after donation from you

MarsTechHAN, 2015-06-20:
Have you ever try to open the Lua's DEBUG option... If I really understand your question, I guess they have simillar function.

TerryE, 2015-06-21:
@MarsTechHAN, Yes, any Lua (firmware) developer could do this, except that (i) is increase the firmware footprint quite a bit, and (ii) it generates a _lot_ of debug output.  Useful for people like me.

However, the Q that we constantly face from Lua developer is "Why to I keep running out of RAM?" or "What is it in my code that is taking up all of the RAM?

What would be very useful for applications programmers is some form of globals, + registry plus closed upvalue walk which tabulates current memory usage so that your average applications programmer an get a handle on this.

vowstar, 2015-06-21:
I'll try to do it today

MarsTechHAN, 2015-06-21:
I agree with you... Lua really need a simple way to debug, especially when this realted to heap consuming

TerryE, 2015-06-21:
@MarsTechHAN, @vowstar, in fact as well as the direct or indirect content of **_G** there are _two_ "hidden" means of context being stored between invocation success Lua callbacks:
  *  Content stored in the Lua registry as is usually the case with data structures used by the Library routines for context. We've had a number of leaks that we've recently fixed related to these, in particular incorrect clean up of bugs in **tmr.alarm()** and **wifi.sta.getap()**.  This is what a `node.getregistry()` would expose.
  *  What Ierusalimschy refers to as "closed up values".  This handling described in detail in Programming in Lua V3, but I've found [this draft paper](http://www.cs.tufts.edu/~nr/cs257/archive/roberto-ierusalimschy/closures-draft.pdf) by Ierusalimschy which cover this in depth in Section 3. And as shown in the following example:
```Lua
--  file a.lua
local T= {1,2,3,4,5,"hello world"}
local function getT() 
  return T 
end
-- T is an open upvalue in getT at this point because T is still in scope
return getT

-- file x.lua
local getT = dofile('a.lua')
-- the outer frame of a.lua of A is now out of scope 
-- however because the closure refers to it, T still magically exists out of site
print (getT()[6]) 
getT = nil
-- Now T is a dead reference and can be GCed 
```
I have enough problems getting my head around this.  It's no wonder that beginners regularly get burnt on it.  Hence my recommendation: avoid using upvalues (also known as outer scoped local variables) in callbacks -- stick to globals.

TerryE, 2015-09-01:
Added to master with #624

closed 2015-09-01 by TerryE