nodemcu/nodemcu-firmware #459: Problem with multiple MQTT subscriptions

devsaurus, 2015-05-31:
I'm setting up an application where the nodemcu firmware is supposed to subscribe to 3 topics as shown here:
```lua
m = mqtt.Client("esp8266", 120, "", "")

m:on("message", function(conn, topic, data) 
  print(topic .. ":" ) 
  if data ~= nil then
    print(data)
  end
end)

m:connect("192.168.0.254", 1883, 0, 0, function(conn) print("connected") end)

m:subscribe("/topic1", 0, nil)
m:subscribe("/topic2", 0, nil)
m:subscribe("/topic3", 0, nil)
```
All seems fine as long as the messages come in asynchronously. But when all three topics deliver messages (more or less) at the same time, the third topic is not processed properly. Eg publishing on another client as follows (srv is on IP 192.168.0.254):
```
mosquitto_pub -h srv -i cli -q 0 -t /topic1 -m 1; mosquitto_pub -h srv -i cli -q 0 -t /topic2 -m 2; mosquitto_pub -h srv -i cli -q 0 -t /topic3 -m 3
```
yields:
```
/topic1:
1
/topic2:
2
/topic2:
2
```
Two messages at once is no problem, no matter which topics are used. But having all 3 in one row causes the last message on the 3rd channel being overwritten by the 2nd message.

Am I doing something wrong here? I'd guess the callback is somehow interlocked, but it seems strange. BTW the other method to subscribe to multiple channles in the README.md doesn't work at all for me:
```lua
m:subscribe({["/topic1"]=0, ["/topic2"]=0, ["/topic3"]=0, nil})
```
It spawns a huge number of prints for a single publish...

TerryE, 2015-05-31:
@devsaurus, this isn't "I found a problem / bug with the firmware"; it's an "I don't know how to get mqtt to work".  This issue list is the former; try www.esp8266.com for the latter and we'll help you there :smile:

drv53b6, 2015-06-01:
how to report spam? this @TerryE  alweys spam with hes www.esp8266.com

TerryE, 2015-06-01:
@drv53b6, esp8266 is my forum and I have no affiliation with it other than it's the only forum that I know of which offers such free support to ESP8266 Lua developers, so I provide help there to developers, just as I do on StackOverflow which doesn't have an active eLua / ESP8266 tag.  Even if you don't agree with my suggesting this developer forum and would recommend a _free_ alternative, my underlying point that this isn't a bug with the firmware still stands.

FourQuarks, 2015-06-01:
@TerryE 

Stop being the unit control man.

marcelstoer, 2015-06-01:
> Stop being the unit control man.

But he's right. The GitHub issue list is not a support forum (even though I'd prefer Stackoverflow over esp8266.com).

drv53b6, 2015-06-01:
@marcelstoer IT NOT GIVE RIGHTS TO SPAM WITH HES FORUM

marcoskirsch, 2015-06-01:
I always wonder what the life of a GitHub troll must be like.

devsaurus, 2015-06-01:
Gentlemen, may I draw your attention to technical aspects.

* My observation above applies to the code in `dev` branch. The instructions given yield that the third message is not shown but the second message is repeated instead.
* When running this example on `master` branch, things work as I'd expect: all three messages are shown correctly.

IMHO the mqtt code in `dev` branch introduced a regression with respect to processing several (>2?) messages in a short time frame. I suspect that commit bcbde08bf70b703e32aa43bdc6a76102a50944d1 is the culprit as it feeds the raw payload buffer to `deliver_publish()`. The code base before this commit seems to be fine for my example.
But this would require a final assessment by the experts, for sure.

drv53b6, 2015-06-02:
ДОРАБОТАЛИ ДО НЕРАБОЧЕГО СОСТЯНИЯ - ВРЕДИТЕЛИ ВРАГИ НАРОДА!

TerryE, 2015-06-02:
@devsaurus, in which case my advice was incorrect and I apologise for this.  It looks as if you may have uncovered a bug in the **mqqt** library.  Terry

devsaurus, 2015-06-02:
@TerryE all is fine. My initial post was more like a FAQ since I posted before having investigated in detail. I take with me that it's advisable to do it vice-versa next time.

Regardless, I'm stuck here since the code appears to be too complex to come up with a PR (these ARE my early steps with MQTT). I hope that someone of the devs can have a look before a potential issue is merged to `master` and reaches a broader audience.

cal101, 2015-06-04:
Moin,

I may have found an issue but didn't tested it.
You may give it a try: 

https://github.com/cal101/nodemcu-firmware/commit/fdcb9823518826cc82f57002f7d47cb01983e6ca

Cal

TerryE, 2015-06-04:
This is plausible since pData is the value parameter passed into `mqtt_socket_received()` and `in_buffer` is working pointer used to step down it.

devsaurus, 2015-06-04:
@cal101 tried your modification on `dev` branch, it appears to solve the issue. The sample code shows all 3 messages with it.
Will you place a PR?

drv53b6, 2015-06-04:
СТАЛИН БЫ ЗА ТАКОЕ -  РАССТРЕЛЯЛ!

TerryE, 2015-06-04:
@drv53b6, Comrade Stalin dead 6 months before I was born and maybe 40 years before you were. Can we please stick to constructive responses?

cal101, 2015-06-04:
Don't feed the troll ...

drv53b6, 2015-06-05:
@TerryE СТАЛИН ЖИВ В НАШИХ СЕРДЦАХ!

devsaurus, 2015-06-05:
Fixed with #465.

closed 2015-06-05 by devsaurus