nodemcu/nodemcu-firmware #461: Always getting nil on getip()

johnjcouture, 2015-06-02:
print("init.lua - Version 4.01 - 06/01/2015 17:38")
majorVer, minorVer, devVer, chipid, flashid, flashsize, flashmode, flashspeed = node.info();
print("NodeMCU "..majorVer.."."..minorVer.."."..devVer)
print("ChipID "..chipid.." FlashID "..flashid)
print("Flash: Size "..flashsize.." Mode "..flashmode.." Speed "..flashspeed)
print("-----------------------------------------------------------------------")

ip = wifi.sta.getip()
print(ip)
--nil
wifi.setmode(wifi.STATION)
wifi.sta.config("mySSID","myPassword")
ip = wifi.sta.getip()
print(ip)
--nil

SOFTWARE VERSIONS =====================================
NodeMCU 0.9.5
ChipID 10546369 FlashID 1261807
Flash: Size 512 Mode 2 Speed 40000000
Using ESP8266 Dev Board (not sure of number, image attached)

[image]

TEST PROCEDURE =================================
* downloaded and used ESPFlasher, (went to Advanced,Restore Default, Operation then Flash) Log indicated that it was done after about a minute.
* downloaded and used LuaLoader 0.87 (very kool program by the way)
* wrote the above .lua script and did an Upload file, bunch of ">" then heap of 21072
* dofile

my cell phone attaches to mySSID ok and the password has been checked.
mySSID is a wifi router that does NOT do DHCP.  That is done by router further
up the ethernet and that works also.

read issues 209,199
Any ideas.

johnjcouture, 2015-06-02:
Image above does not have DIP 8 (GPIO0) set ON but when I programmed it, I was careful to set and unset it.  For those interested, this is a schematic of the board http://www.bpower.nl/pic/ESP-201-DEV-bw.png

TerryE, 2015-06-02:
Your application is pretty much doing what I would expect it to do.  I don't think that this is a bug with the firmware per se, so this issue doesn't belong here.

You either need to connect or autoconnect to tell ESPCONN to negotiate the IP and you need to wait until it has a chance to so to.  The typical way to do this is to set up  tmr.alarm() to check the getIP on, say a 100mSec poll loop and once it has a valid IP return, then stop the timer and continue with the rest of execution.  If this works, then please close this issue.

johnjcouture, 2015-06-02:
@TerryE

Mr. Ellison,

Thank you for your quick reply.

In the updated code below, it is interesting that it prints **the second nil**
before the hello world.  

Also it does not return with a ">" prompt after
the hello world but maybe that is because one must
put the cr/lf in the print statement.

Finally, I guess I am in the wrong place because I did not 
mean to imply that it was a bug but rather I was trying to 
determine if I made a mistake!  :bowtie:

If you could direct me to the appropriate forum, I would appreciate it.

======================================
After it prints "Flash: Size ..."
nil
The IP for this module is;
nil
> hello world

======================================
Updated Code:
======================================

    print("init.lua - Version 4.03 - 06/02/2015 07:15")
    majorVer, minorVer, devVer, chipid, flashid, flashsize, flashmode, flashspeed = node.info();
    print("NodeMCU "..majorVer.."."..minorVer.."."..devVer)
    print("ChipID "..chipid.." FlashID "..flashid)
    print("Flash: Size "..flashsize.." Mode "..flashmode.." Speed "..flashspeed)
    print("-----------------------------------------------------------------------")
    
    ip = wifi.sta.getip()
    print(ip)
    --nil
    wifi.setmode(wifi.STATION)
    wifi.sta.config("mySSID","myPassword")
    
    -- Insert a delay to allow module to get the IP
    tmr.alarm(0, 1000, 0, function() print("hello world") end )
    
    -- Now check to see if it has an IP
    ip = wifi.sta.getip()
    print("The IP for this module is: ")
    print(ip)
    
======================================

johnjcouture, 2015-06-02:
@TerryE

Mr. Ellison,

Thank you for your quick reply.

In the revised code below, it is interesting that it prints **the second nil**
before the hello world.  

Also it does not return with a ">" prompt after
the hello world but maybe that is because one must
put the cr/lf in the print statement.

Finally, I guess I am in the wrong place because I did not 
mean to imply that it was a bug but rather I was trying to 
determine if I made a mistake!  :bowtie:

If you could direct me to the appropriate forum, I would appreciate it.

======================================
After it prints "Flash: Size ..."
nil
The IP for this module is;
nil
> hello world

======================================
Updated Code:
======================================

    print("init.lua - Version 4.03 - 06/02/2015 07:15")
    majorVer, minorVer, devVer, chipid, flashid, flashsize, flashmode, flashspeed = node.info();
    print("NodeMCU "..majorVer.."."..minorVer.."."..devVer)
    print("ChipID "..chipid.." FlashID "..flashid)
    print("Flash: Size "..flashsize.." Mode "..flashmode.." Speed "..flashspeed)
    print("-----------------------------------------------------------------------")
    
    ip = wifi.sta.getip()
    print(ip)
    --nil
    wifi.setmode(wifi.STATION)
    wifi.sta.config("mySSID","myPassword")
    
    -- Insert a delay to allow module to get the IP
    tmr.alarm(0, 1000, 0, function() print("hello world") end )
    
    -- Now check to see if it has an IP
    ip = wifi.sta.getip()
    print("The IP for this module is: ")
    print(ip)
    
======================================

TerryE, 2015-06-02:
John, the name is Terry :smile:.  Have a read of my [nodeMCU Unofficial FAQ](http://www.esp8266.com/wiki/doku.php?id=nodemcu-unofficial-faq).  Sorry, it's a bit heavy, but this will explain a lot of what you need to understand. There's an [ESP8266 forum](http://www.esp8266.com/viewforum.php?f=17) which provides free advice to application developers using this firmware as well as the [nodeMCU forum](http://bbs.nodemcu.com/) though a good working knowledge of Chinese helps on this one.

johnjcouture, 2015-06-02:
@TerryE 

Thank you Terry!  I went through the FAQ (nice reading).  I was also reading through your explanation on wifi.sta.autoconnect(1) in the ESP8266 forum from a couple of days ago.  Mine doesn't work with that either.  I'll go back and look at that alarm function.

Anyway, Thank you again for your help I'll close this issue and start working through the ESP8266 Community Forum.

closed 2015-06-02 by johnjcouture

weedmanu, 2016-07-25:
hello,
just that for wait connection : 
tmr.alarm(0, 1000, 1, function()
   if wifi.sta.getip() == nil then
      print("Connecting to AP...")
   else
      print('IP: ',wifi.sta.getip())
      tmr.stop(0)
   end
end)

shuvamiitkgp89, 2017-05-26:
i am facing the same problem ... I have tried with the code 
hello,
just that for wait connection :
tmr.alarm(0, 1000, 1, function()
if wifi.sta.getip() == nil then
print("Connecting to AP...")
else
print('IP: ',wifi.sta.getip())
tmr.stop(0)
end
end)
 but the "Connecting to AP..." is continuing ... not getting the static IP address ... is it the problem with the module?

weedmanu, 2017-06-06:
Finally, I gave up LUA for the arduino IDE, it's better for me because I already know, it saves me from having to learn a new language