nodemcu/nodemcu-firmware #468: Additional node.compile_options to limit generated debug info and save code space

TerryE, 2015-06-06:
This is just a thought for comment. 

We know that node.compile() reduces the size of Lua bytecode typically by the order of 40%.  So why limit this to this dump function?  Why not have the option to turn off creation of debug information during the compile process? This would allow us to load `.lua` source files with the same code saving as `.lc` bytecode ones.

There is also a sensible intermediate here, and that is only function names and line number information is used in error reporting.  So why not allow a half-way house of not including local and upvalue names (as these are only used by the `debug` module anyway, which isn't currently implemented)?  

Also the mechanism for storing line numbers is horribly space inefficient (a 32-bit integer per bytecode).  It would be trivial to use some offset encoding which would get this down to one byte per source line.  OK, this would mean that error reporting would have to walk this byte array to calculate the line number, but this loop only needs to be executed on error and most eLua functions only have a couple of dozen lines at most so this isn't a burdon.  With this option, you'd still get error reporting but with a 35% (say) code saving instead of 40%.

These sorts of saving would be _very_ worth having.

cal101, 2015-06-06:
+1 

having stacktraces on errors in callbacks helped me debug my lua code.

TerryE, 2015-06-06:
This would increase compile times slightly since it would add an extra "consolidation" pass (because it's not just the debugger that uses these meta tables; the compiler also uses this data during compile to resolve upvalues, etc., but if a second pass is the cost of adding effectively 50% to your RAM, then it might be an option that a lot of users would consider.

jorgempy, 2015-06-07:
+1. Awesome!

vowstar, 2015-06-08:
Cool idea

TerryE, 2015-06-09:
The main use of the line info during execution is during error reporting.  There's a macro in `ldebug.h`
```C
#define getline(f,pc)	(((f)->lineinfo) ? (f)->lineinfo[pc] : 0)
```
and a short wrapping routine `int currentline(L,fn)` in `ldebug.c`.  Hence each bytecode instruction is a 32bit word  and has an associated instruction->line-no map which is also a  32bit word -- a 100% overhead.  My thinking is to use a runlength encoding scheme, e.g. variable (typically one) byte for source line e.g.  of the format `0b0fnnnnnn` or two bytes `0b1fnnnnnn nnnnnnnn where the `n...n` is an unsigned integer, typically 0..63 but can be 0..32767 and the flag bit `b`= 0 if there are `n` instructions in this line or lag bit `b`= 1 if there are `n` lines with no instructions.  The first line number is an absolute variable encoded start value. 

I've just done a luac on a short 13 line 41 instruction function which starts at line 2, and has 5,5,0,0,6,3,13,2,0,0,6,1 instructions per line.  Under this scheme, the lineinfo would be encoded:
```
02 05 05 42 06 03 0D 42 06 01 00
```
that is 11 bytes instead of 164. OK doing a getline() would now be a function call which will involve perhaps 7x15 instruction to walk down to get the line number of pc=28, say.  Big deal.  An extra few µs or so processing an error or so to reduce the instruction lookup vector by 20x.  I could take 10 times this number of instructions to do this and it still wouldn't be material.

TerryE, 2015-06-29:
I have started doing this work.  I will first do it on a standard eLua config running on my PC -- simply because `gdb` is available there and then port it to nodeMCU and test it locally on my dev system before push it to my fork.  The patch is turning out to be pretty straight forward.

TerryE, 2015-07-13:
I have this code working at least for the luac.cross build.  I should have a tested ESP8266 version in the next week or so (as there isn't a cut down eLua version which attached to say a POSIX os interface as its "native" version for host testing. :disappointed: 

One thing that I have noticed I that Lua uses a doubling strategy for allocating growing resources (such as table hashes, code and line pointer vectors, so compiling a ten line function allocates 1,2,4,8,16 element vectors and then cuts them back to 10 (e.g. 6 allocation calls) during the `lparser.c:close_func()` wind-up.  However, `lua.c:470` does a `legc_set_mode( L, EGC_ALWAYS, 4096 )` which means that `luaC_fullgc()` is called *on every alloc call* and this is very expensive. Surely `EGC_ON_ALLOC_FAILURE | EGC_ON_MEM_LIMIT` would be a better option?

I avoid this in my routine by doing the same as we did in PHP OPcache, which is to use a two pass approach with the first pass just calculating the exact size. This extra pass might seem like overkill but this a lot less work than 5+ extra GC calls.

TerryE, 2015-07-25:
I have this patch working pretty solidly on standard Lua, but I still have a few errors being thrown up by the [Lua 5.1 test suite](http://lua-users.org/lists/lua-l/2006-03/msg00716.html), for example compiling the invalid source:

```Lua
 a=print+;
```

creates an error path which results in a 32 byte memory leak. But getting there

jorgempy, 2015-07-27:
thanks for the update @TerryE ! good stuff!

TerryE, 2015-08-03:
Sorry I am in the middle of building myself a house (which is why I got into HA / IoT / ESP8266 in the first place) s this bug hunting has been going slowly.  The test suite is brilliant at exposing abstruse bugs.  For example the test harness exposes its one interface and in the API tests it repeats the tests over and over again ramping up the "maximum" memory by three bytes at time to force "out of memory" throws at just about every point in the code where they could possibly happen. For example the `lfunc.c:luaF_newproto()``contains the lines

```C
  Proto *f = luaM_new(L, Proto);
  // ... 
  f->lineinfo.unpacked = luaM_new(L, UnpackedLine);
```
and the corresponding `lfunc.c:luaF_freeproto()``contains the lines

```C
  if (f->lineinfo.unpacked) {
    // ...
    luaM_freearray(L, f->lineinfo.unpacked->info, f->lineinfo.unpacked->size, int);
    // ...
  }
  //  ...
  luaM_free(L, f);
```
However the `luaM_new(L, UnpackedLine)` can throw a `LUA_ERRMEM` which is caught up the stack and recovered with a garbage collection, but this leaves `f->lineinfo.unpacked` with the default `0xabababab` which is non-zero and so the clear-up code attempts to free this.  Segfault.  Fun, fun.  Still another one down.

jmattsson, 2015-08-06:
Looking forward to seeing the results of your efforts, @TerryE. It sounds like we can look forward to some more stable node.compile() runs too then.

And color me impressed that you're finding time for this while building a house!

TerryE, 2015-08-19:
For those interested, I've now got this patch working on standard Lua with a 100% pass on the strictest setting of the Lua 5.1 test suite.  It's here on my [github lua fork](/TerryE/lua).  This was a pretty good intro into the internals of the Lua parser / code generator.  Next steps are to extend the test suite to cover the additional patch functionality, and then to start the port to nodeMCU.  I am not going via eLua.  This will have to be a back port if there's the traction, because I don't have a decent debug platform to test standard eLua images.

This wider understanding has thrown up a number issues around the compiler -- e.g. it really thrashes dynamic memory with malloc / frees.  Need to consider if there are any benefits to taking a look at this next.  Though my preference would be a slicker way of using luac.cross + prebuilding SPIFFS images for true embedded configs.  Watch this space and sorry this is taking so long.

jmattsson, 2015-08-19:
Nice going!

With your knowledge of the Lua internals, do you see *any* possibility of being able to execute Lua bytecode directly from flash without having to first load it into RAM? From what little I've poked around it seemed rather unlikely, but it would be an amazing RAM saving if it could be achieved. For the sake of the argument, feel free to ignore details such as the .lc file data not being available in a contiguous block.

TerryE, 2015-08-19:
@jmattsson Johnny, have a look at `lundump.c` which is the bytecode loader.  The issue isn't just contiguity, but also guaranteeing that a mapped vector isn't updated or deleted.  If we were using a ROMFS instead of SPIFS (or even a unified ROMFS + SPIFS) where the ROMFS it part of the flash load, then it would be straight forward to map the "ROM" version of the `Proto:code` and include extra logic in the free routine to ignore free requests to ROM.  I also assume that such modules would always be fully stripped.  Even if we did this, you've still got a lot of dirty objects that will need to be clone to RAM: the Prototype header, all of the data element, constants, etc.  This is analogous to how PHP uses its OPcache.    

Where this might help is for "core" Lua modules that are in effect almost part of the firmware build. Worth putting on the "to consider" list, but there are other low hanging fruit to go after before this. :smile:

jmattsson, 2015-08-20:
Thanks for that Terry! I'll keep that in mind in case we find ~~an excuse~~ a need for it here at work! :D

TerryE, 2015-09-03:
Just got clean compiles for my nodemcu master fork which applies this patch with and without `LUA_OPTIMIZE_DEBUG` set.  Testing starts tomorrow.  Once I've done a first pass and if it looks OK, then I'll upload this to my github remote.  I'll need at least one independent tester to check it out before I issue a pull request.

TerryE, 2015-09-03:
OK, I have just uploaded this to my noduleMCU fork on this [stripdebug](/TerryE/nodemcu-firmware/tree/stripdebug) branch.  

It adds an extra `node.stripdebug([level[, function]])` command where  

*    level
     * 1. don't discard debug
     * 2. discard Local and Upvalue debug info
     * 3.  discard Local, Upvalue and lineno debug info.
*    function: Function to be stripped as per setfenv except 0 not permitted. 
 
If no arguments are given then the current default setting is returned.  If function is omitted, this is the default setting for future compiles.  The function argument uses the same rules as for `setfenv()`.

You need to define `LUA_OPTIMIZE_DEBUG` to your default strip level.  Leaving it undefined or 0 effectively removes the patch at the C preprocessor level. 

Here is a little test to show the heap savings.  Essentially you get all of the benefits of doing a node compile in incremental compilation with level 3 and 95% of the savings, but still get line info in your errors at level 2.

```Lua
-- test.lua
local opt=...
node.stripdebug(opt)
do
  for i = 1,3 do collectgarbage() end
  print(node.heap())
  local a={}
  for i,k in ipairs {"file1","file2", "file3"} do
    a[i] = loadfile (k .. ".lua")
  end
  for i = 1,3 do collectgarbage() end
  print(node.heap())
end

TerryE, 2015-09-08:
For anyone reviewing this, [here](http://files.ellisons.org.uk/esp8266/NodeMCU-LCD.pdf) is a draft of my working paper with documents the patch.  Sorry but this is a PDF print of my local dokuwiki copy a I have my editor access to the esp8266 withdrawn  (probably just a temporary hiccup).

TerryE, 2015-11-02:
Implemented and being promoted to `dev`.

closed 2015-11-02 by TerryE