nodemcu/nodemcu-firmware #475: Base64 lua module is broken

Cougar, 2015-06-08:
I found that in many cases base64 from included module https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_modules/base64/base64.lua commited by @AllAboutEE (probably from http://lua-users.org/wiki/BaseSixtyFour) and other libs give different results.

After some trial and error tests I found that the lua implementation (especially decoder) is totally broken.

  > = base64.enc("DDD")
  RERE
  > = base64.enc("HHH")
  RERE

 It can't even decode its own encoded data.

  > = base64.dec(base64.enc("test string"))
  :29:9:9473

These are correct results:

  > echo -n "DDD" | base64 
  RERE
  > echo -n "HHH" | base64 
  SEhI
  > echo -n "test string" | base64 | base64 -d
  test string

The ```crypto.toBase64()``` works fine for encoding.

  > = crypto.toBase64("DDD")
  RERE
  > = crypto.toBase64("HHH")
  SEhI

For decoding I use https://github.com/ErnieE5/ee5_base64 module right now.

  > base64 = require("ee5_base64")
  > = base64.decode(base64.encode("test string"))
  test string

creationix, 2015-06-09:
For what it's worth, the new crypto module in the dev branch has a native base64 implementation in C.

Cougar, 2015-06-09:
But only encoding right now (crypto.toBase64()) or am I blind to find a decoder ;-)

cal101, 2015-06-10:
What do you think about changing the method names to a more common base64Encode/base64Decode?

creationix, 2015-06-10:
@Cougar I didn't implement decoding since it's already in the code for ssl. (also again in the ROM, but @cal101 knows that is somewhat tricky to use)

@cal101 I don't have strong opinions on the names.  My background isn't C so I don't know what names are common there.

mkevins, 2015-09-02:
@creationix could you give a quick example how to use decoder? I can't seem to figure it out on my own...

TerryE, 2015-12-22:
We need to delete this example when I implement the C library version.  If someone wants do dig it out of the archive and use it, that's up to them, but we shouldn't include it in the distro.  I've assigned this to me as a reminder.

closed 2016-03-02 by TerryE