nodemcu/nodemcu-firmware #476: help with connecting antenna to ESP8266-08

joefly888, 2015-06-09:
First, I see you guys all hard at work developing the firmware on this relatively new and promising product.  Thank you for your initiative..

Second, I apologize as I know this is not a firmware issue, but I have looked for solution all over and no response.  So I was hoping the more knowledgeable people here would be able to help. 

I inadvertently bought multple ESP8266-08 http://www.ebay.com/itm/291419560511?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT looking for the most compact small version, not realizing that this does not include an antenna pin/pad. 

Does anybody know how to attach an antenna to this version of ESP8266?  Again, I apologize for the inappropriate post but hopefully someone here can help..

TerryE, 2015-06-09:
Joe, this is definitely not an issue for here: it's not a bug in the nodeMCU firmware; it's not even about this firmware -- so please close this call.

This is a sealed module.  Surgery on it is not for the faint hearted and might well be impossible.  The cost $4.  Use them as prototypes or resell on ebay; read the data sheets and select the correct module next time.

nrbrt, 2015-06-09:
While I agree with Terry about the off-topic nature of this question, I would try the small connector-pad on the short edge of the module, near the REST connector. Surgery and removal of the shield should not be necessary.

TerryE, 2015-06-09:
Early pin-outs show the ext ant connector where rest is so I suspect that the separate pin as shown on your pic is this connector.  *Please close this issue.*

joefly888, 2015-06-09:
Apologize again for interruption. But thanks for comments. Please close issue.

closed 2015-06-09 by joefly888

TerryE, 2015-06-09:
That's fine Joe and thanks for closing this.  You can always get support on the [nodeMCU forum at esp8266.com](http://www.esp8266.com/viewforum.php?f=17).