nodemcu/nodemcu-firmware #477: Using rboot as bootloadern (OTA support)

jorgempy, 2015-06-10:
Hi, i wonder if some of the people that knows the build process of nodemcu have seen [rboot bootloader](https://github.com/raburton/esp8266/tree/master/rboot), if so, it sounds like a good option to achieve OTA upgrades, a simple wrapper of the OTA api exposed by rboot will allow us to upgrade the esp's easily.
I read in another issue that the problem was that the nodemcu image is bigger than the rom boundary (512kb if i recall correctly). The thing with this bootloader is that it allows roms to have arbitrary sizes, from the readme of rboot:
```  
It has the following advantages over the Espressif loader:
  - Open source (written in C).
  - Supports up to 256 roms.
  - Roms can be variable size.
  - Able to test multiple roms to find a valid backup (without resetting).
  - Flash layout can be changed on the fly (with care and appropriately linked
    rom images).
  - GPIO support for rom selection.
  - Reserves less ram (16 bytes vs. 144 bytes for the SDK loader).
  - Documented config structure to allow easy editing from user code.
 [...]
You can can then write your two roms to flash
addresses 0x2000 and (half chip size + 0x2000). E.g. for 8Mbit flash:
  esptool.py -fs 8m 0x0000 rboot.bin 0x2000 user1.bin 0x82000 user2.bin
```
is this a possibility? is it possible to support a 'multi-bootloader' makefile or something like that?
if so, i would gladly write the lua wrapper for rboot OTA. unfortunately, im not that familiar with the esp build chain to be sure on the makefile front.

creationix, 2015-06-10:
+1 this would be cool.  Also rboot saves 128 bytes of heap.  Not much, but every little bit of savings is a good thing when we are so constrained.

dnc40085, 2015-06-17:
+1 OTA updates would be nice and more memory is always a good thing no matter how small the gain is.

TerryE, 2015-11-02:
OTA might be nice but the current contributors won't be implementing this.  If anyone wishes to offer an implementation, then please reopen this issue.

closed 2015-11-02 by TerryE