nodemcu/nodemcu-firmware #479: Interrupts: "down" edge behaves as "both"

Suxsem, 2015-06-13:
Hi. I have just tested this code on my esp8266 01:
```
gpio.mode(4,gpio.INT)

function pincb(level)     
    print("interr!")
end

gpio.trig(4,"down",pincb)
```

Well...i get the "interr!" debug string on RISING AND FALLING edges. As if I put "both".
No problem with "up" edge!

Can you test please?

Suxsem, 2015-06-15:
No one?

znsoft, 2015-06-15:
its a contact bounce (relay chatter; relay flutter), try this tmr.delay(700000) after print() work fine!
 i not found english doc but there is on russian: https://ru.wikipedia.org/wiki/%C4%F0%E5%E1%E5%E7%E3_%EA%EE%ED%F2%E0%EA%F2%EE%E2

marcelstoer, 2015-06-16:
Good information about the debounce subject is here: http://esp8266.co.uk/tutorials/introduction-to-the-gpio-api/#debouncing

Suxsem, 2015-06-16:
Thanks... I found it odd because with "up" i don't have this problem...

TerryE, 2015-06-16:
Bounce is usually a electrical manifestation a mechanical mechanism such as a relay.  You will get mechanical bounce at the relay-open side but usually this won't give rise to false signals.  

Whatever the issue, it's nothing to do with nodeMCU.  Please close this call. Thank-you.

Suxsem, 2015-06-16:
Ok, thank you :)

closed 2015-06-16 by Suxsem