nodemcu/nodemcu-firmware #480: TMR.WDCLR() LEAKS MEMORY WHEN CALLED TOO OFTEN

Loooopy, 2015-06-14:
When calling tmr.wdclr() to often it will leak memory.

Memory Leaking code:
    function forwardandrewind()
        if(forward == 1) then
           forward = 0
           gpio.write(3,gpio.HIGH)
        else
           forward = 1
           gpio.write(3,gpio.LOW)
        end

       for x=1,16000 do
           gpio.write(4,gpio.HIGH)
           tmr.delay(1)
           gpio.write(4,gpio.LOW)
           tmr.delay(1)
           tmr.wdclr()
      end

      print(node.heap())
    end

Code that do not leak:
    function forwardandrewind()
        if(forward == 1) then
           forward = 0
           gpio.write(3,gpio.HIGH)
        else
           forward = 1
           gpio.write(3,gpio.LOW)
        end

       for x=1,16000 do
           gpio.write(4,gpio.HIGH)
           tmr.delay(1)
           gpio.write(4,gpio.LOW)
           tmr.delay(1)
           if(x%1000) then
               tmr.wdclr()
           end
      end

      print(node.heap())
    end

Here i substitute the 
    tmr.wdclr()
with
   if(x%1000) then
      tmr.wdclr()
   end

TerryE, 2015-06-14:
Not sure how this could cause a leak.  The `tmr.wdclr()` function simply does the following
```C
static int tmr_wdclr( lua_State* L )
{
  WRITE_PERI_REG(0x60000914, 0x73);
  return 0;  
}
```
This is a write to the hardware wdclr register.  There's no scope for memory leak here.  You are breaking Espressif coding guidelines by sitting in a task this way.  Use the and pcm library module which calls the  PWM driver if you want to use your ESP8266 for PWM.

Loooopy, 2015-06-15:
Thanks for the fast reply!

Do it matter if i used the pcm library or made a special case, i think i should be able to call it that way without a memory leak? 

Have you tested the code and can you explain why it leaks memory if its not the tmr.wdclr()?

>You are also breaking Espressif coding guidelines by sitting in a task this way. 

This i know but if I do not expect any communication it should be okay, but i know I break the guideline but this is not the issue here its the call to tmr.wdclr().

>Use the pcm library module which calls the PWM driver if you want to use your ESP8266 for PWM.

Can you use the PWM driver to drive a stepper motor exact amount of steps, it looks like it only support a start and stop and i have to handle that manually, when i can't grantee that it moved lets say 20 steps.

closed 2015-08-24 by TerryE