nodemcu/nodemcu-firmware #482: net.createConnection() return nil

vudangngoc, 2015-06-14:
I build latest code from both dev and dev096 and copy code from example but it doesn't work as expected 
```
> local conn = net.createConnection(net.TCP, 0)
> print(conn)
nil

```

TerryE, 2015-06-14:
What's the firmware bug?  Use the [nodeMCU forum](http://bbs.nodemcu.com/) or the English [ESP8266.com forum](http://www.esp8266.com/viewforum.php?f=17) for applications support.

vudangngoc, 2015-06-15:
I check this bug with branchs dev and dev096, both of its have this issue

TerryE, 2015-06-15:
`createConnection()` fails if your WiFi isn't connected.

vudangngoc, 2015-06-15:
I think my MCU connected to wifi
[image]

pastukhov, 2015-06-15:
```lua
sk=net.createConnection(net.TCP, 0)
    sk:on("receive", function(sck, c) print(c) end )
    sk:connect(80,"192.168.0.66")
    sk:send(
"GET / HTTP/1.1\r\nHost: 192.168.0.66\r\nConnection: keep-alive\r\nAccept: */*\r\n\r\n"
    )
```
from http://www.nodemcu.com/docs/net-socket-module/#net-socket-module-connect

vudangngoc, 2015-06-16:
You can see my code is similar with first line of your example, but it doesn't work as my screenshot

pastukhov, 2015-06-16:
But where is```    sk:connect(80,"192.168.0.66")```?

vudangngoc, 2015-06-16:
net.createConnection(net.TCP, 0) return nil so that I cannot call sk:connect(80,"192.168.0.66")

jorgempy, 2015-06-17:
i believe the problem is the 'local' keyword. i appears you are trying it in the repl, in the repl each line has it's own scope, so a local var is nil in the next line. You should try either not using local or putting it in a file a running it using dofile.

TerryE, 2015-06-17:
> in the repl, in the repl each line has it's own scope, so a local var is nil in the next line
@jorgempy,   I was puzzled abut this bug. Thanks for pointing out what should have been obvious.  However, one subtle amendment, it is syntactically complete chunks that Lua executes in "-i" mode.  You can tell when the chunk isn't complete because you get the `>>` prompt instead of the `>` one.

vudangngoc, 2015-06-18:
the local keyword make conn variable have scope within one code line. I put all code to do - end block and it work. Thank you very much!

[image]

closed 2015-06-18 by vudangngoc

jorgempy, 2015-06-18:
@TerryE yes, chunks... thanks for the correction, and for the do .. end suggestion ;)