nodemcu/nodemcu-firmware #487: Neopixel library no longer works properly.

creationix, 2015-06-18:
Some time recently (not sure, maybe a few couple weeks ago), the dev or dev096 branch (and now master) stopped working for my neopixels (ws2812).

It mostly works, but the first pixel will light up bright green every other call to `ws2812:write` or `ws2812:writergb`.

I have to work around this by sending just the first pixel after sending a full refresh.  Sometimes it still flickers visibly because the two lua calls are slow enough.

Is anyone else seeing this?

creationix, 2015-06-18:
At first I thought I broke my pixels, but I've tried several different strips and they all act the same.

vowstar, 2015-06-18:
I think I'll buy some ws2812 for testing ... :D

creationix, 2015-06-18:
@vowstar for what it's worth, I'm testing with two different kinds.

The newer ones like http://www.adafruit.com/products/1487 seem to have a slightly different protocol than the older ones like http://www.adafruit.com/products/1734 (rgb vs grb ordering)

jorgempy, 2015-06-18:
i am seeing this as well, it is very strange.

MarsTechHAN, 2015-06-19:
We try to make the program work : )
Wish you can sucessfully remote your WS2812

creationix, 2015-06-26:
I think this may be related to printing the `>` on the serial console.
When using the serial console to send commands if I prefix the line with a single `"\0\0\0"` write and then do the real write, the colors are correct.  I think the first tiny write will reset whatever bad state the neopixel is in.

But if I do this in two lines, it doesn't help.  It has to be one command.

This will result in the first pixel being wrong with 255 for the first byte.
```
> ws2812.write(3, "\0\0\0")
> ws2812.write(3, "\1\2\3\4\5\6")
```

But this will work
```
ws2812.write(3, "\0\0\0")ws2812.write(3, "\1\2\3\4\5\6")
```

kbeckmann, 2015-07-15:
Has anyone else tried the new code? Feel free to close this issue if the issue is solved.

creationix, 2015-07-16:
I'm having a hard time getting this to work with my ws2811 neopixels.  They worked before with the old ws2812 code except red and green were swapped compared to my ws2812 neopixels. (This made the `ws2812.write` function very convenient)

creationix, 2015-07-16:
Hmm, not working with my ws2812 neopixels either (the adafruit 8x8 array).  Am I doing something wrong.  I'm testing with the nodemcu devboard 1.0 flashed in dio mode with latest esptool.py.

creationix, 2015-07-16:
Part of the problem was my ws2812 has some bad wires.  I did get the ws2811 working by slightly slowing down the timing to the recommended values.  I'll test again with unmodified code now that I figured out my wiring issue.

creationix, 2015-07-16:
Yep, confirmed.  The ws2812 works with the current code as long as I wire power, ground and data-in properly.  But the ws2811 (also branded as neopixels on adafruit) only works if you slow down the timings.

```diff
diff --git a/app/modules/ws2812.c b/app/modules/ws2812.c
index de45fca..5a79124 100644
--- a/app/modules/ws2812.c
+++ b/app/modules/ws2812.c
@@ -31,9 +31,9 @@ static void ICACHE_RAM_ATTR ws2812_write(uint8_t pin, uint8_t *pixels, uint32_t
   pixel = *p++;
   mask = 0x80;
   start_time = 0;
-  t0h  = (1000 * system_get_cpu_freq()) / 3333;  // 0.30us (spec=0.35 +- 0.15)
-  t1h  = (1000 * system_get_cpu_freq()) / 1666;  // 0.60us (spec=0.70 +- 0.15)
-  ttot = (1000 * system_get_cpu_freq()) /  800;  // 1.25us (MUST be >= 1.25)
+  t0h  = (1000 * system_get_cpu_freq()) / 2857;  // 0.35us (spec=0.35 +- 0.15)
+  t1h  = (1000 * system_get_cpu_freq()) / 1429;  // 0.70us (spec=0.70 +- 0.15)
+  ttot = (1000 * system_get_cpu_freq()) /  667;  // 1.50us (MUST be >= 1.25)
 
   while (true) {
     if (pixel & mask) {
```

kbeckmann, 2015-07-16:
Thanks for the great feedback @creationix ! Would it make sense to have an optional argument that lets the user use ttot=1.25us? Some users might need the high performance.

creationix, 2015-07-16:
maybe we could add a ws8211 module that shares code with ws8212 but doesn't convert grb to rgb and uses a slightly slower delay?  They are different protocols after all, just very, *very* close.

kbeckmann, 2015-07-16:
Good idea, I'll was thinking the same and I'll start coding on it. I hope the footprint won't become much larger, it's basically just a couple of short strings and function pointers.

marcelstoer, 2015-07-16:
> maybe we could add a ws8211 module that shares code with ws8212

How would structure that? Would there be a common ws821x module which the other two depend upon or would the shared code be outside a module? I'm asking because in #386 I asked about inter-module dependencies.

creationix, 2015-07-16:
I don't think you'd want a common module at the lua layer, but there could be a common C file that get's included if any of the three are used.

kbeckmann, 2015-07-16:
Right. I was thinking to add app/driver/ws281x.c and call it from app/module/ws2811.c and app/module/ws2812.c with different arguments.

robertfoss, 2015-07-16:
or maybe add it all into ws281x and have both calls defined there?

kbeckmann, 2015-07-16:
I think it's a question about API compatibility as well. It sucks to break it so I think we need to keep the ws2812 module intact.

Also thinking about that the ws2812.write() call uses GRB. It would however be nice to have a harmonised API so we get the same result when doing the following:

ws2801.write(3, 4, string.char(255, 0, 0))
ws2811.write(3, string.char(255, 0, 0))
ws2812.write(3, string.char(255, 0, 0))
-- The first led should turn red.

But since ws2812.write() is GRB we'll break compatibility with old code so that's probably a bad idea.

So how about I move the bitbang-code to app/drivers/ws281x.c and add a new module, ws2811?

robertfoss, 2015-07-16:
That sounds good enough for me. Let's make sure to disable ws281x.c if
!defined(USE_WS2811) && !defined(USE_WS2812).

kbeckmann, 2015-07-16:
https://github.com/kbeckmann/nodemcu-firmware/commit/32c394c5a6a683489847b710e80e951c5af252ea

kbeckmann, 2015-07-17:
Works fine and have verified with my scope that the correct times are used.. No green pixels and ws2811.write() is indeed 20% slower than ws2812.write().

@robertfoss where should I put that code?

robertfoss, 2015-07-17:
I meant, just don't compile/link ws281x.c into builds where it isn't used.

creationix, 2015-07-17:
@kbeckmann one note for the ws2811 driver.  Simply making `ttot` larger was not enough.  I has to make the other values slower too.  See my diff above.

kbeckmann, 2015-07-17:
@creationix Right. I increased the other values to your values as well. They don't change the total time of the pulse since they only change the pulse width. 

The values I picked before was something that simply worked fine, but using the values proposed by the datasheet work fine as well.

kbeckmann, 2015-07-17:
@robertfoss Maybe I'm naïve but shouldn't the linker get rid of dead code :)?

robertfoss, 2015-07-17:
That's the best case scenario? How are the other drivers compiled/linked?

kbeckmann, 2015-07-17:
It makes sense to have ifdefs.. Reduces build time and guarantees that the code isn't included in the linked binary.
Added that to my branch and it works fine. https://github.com/kbeckmann/nodemcu-firmware/commit/638a7278527eeda8122f38661eab90a2a667618b

Feel free to give me more comments and I'll fix it. I'll make a pull request tomorrow.

rjharmon, 2015-08-30:
Hah, I thought I'd fried the first element of my pixel-strip by overheating during soldering, but the dev branch works well.  Thanks all

devsaurus, 2015-12-13:
Closing this since WS2812 module is supposed to be working fine.

closed 2015-12-13 by devsaurus

karlp, 2015-12-22:
Confirmed fixed on the _dev_ branch, I logged in to comment that it was still borked on latest master.

kbeckmann, 2015-12-22:
Oh it hasn't been merged to master yet? Maybe @marcelstoer can have a look and pull in  c8d1cacd243496f43be55e3a33d4176e7fcdbe73 to master.

TerryE, 2015-12-22:
It master is broken then we might consider a merge, but what we aim to do is to rebaseline dev to master at sensible checkpoints.

kbeckmann, 2015-12-22:
@TerryE Yes that sounds good but i realized after posting that my fix was indeed merged to master a while ago. I could have a look at it, build master and see if my leds work.

TerryE, 2015-12-22:
@kbeckmann, Konrad you know some variation on the old joke an Englishman, an Australian and a Swiss guy walk into a bar ... Well that's us. There only a small group doing the work on the Lua core, we've only got limited hobby time to spend on this so we have to prioritise. We don't and to get stucked into complicated release procedures, because this would just bring the real core development and bug fixing to a standstill. Sorry.