nodemcu/nodemcu-firmware #493: How to test Smartconfig?

adoph, 2015-06-20:
I tried using smartconfig, but has been unable to succeed. I can not find the information to solve smartconfig. I know there are many versions of ESP-Touch. I'm not sure the problem is nodemcu or  App. Can you provide information to help me?

MarsTechHAN, 2015-06-21:
One of the issue is... fix the smartconfig

So... now we are taking step on it...But we have not finished...
If you want to use ESPTOUCH, you can make your own bin，
look it up->https://github.com/obuil/NodeMcu

vowstar, 2015-06-21:
Waiting our v1.1.2 branch, we will import esptouch and use espressif's open sourced app.

adoph, 2015-06-21:
I look forward to your new version. Thank you.

Air000, 2015-06-30:
if the v1.1.2 branch support AirKiss?

MarsTechHAN, 2015-06-30:
@Air000 Yes

vowstar, 2015-07-03:
@Air000 airkiss is OK, test v0.9.6 branch and see documents.

vowstar, 2015-07-03:
Smart config is OK. Close #493.

pastukhov, 2015-07-06:
i'm getting 
```
> PANIC: unprotected error in call to Lua API (stdin:3: bad argument #2 to 'format' (no value))
```
with wifi.smartstart example from documentation on  today v0.9.6

Air000, 2015-07-07:
@vowstar I test the smartconfig with firmware 0.9.6-dev_20150704. But fail with reboot:

> dofile("smartconfig.lua")                                 
> Success. SSID:Terry3G ; PASSWORD:msjjsns                  
bB��ʦ�Sb9�1�)]�1�)�!�N�hC                                   
NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4          
lua: cannot open init.lua

Air000, 2015-07-07:
And I compile a dev112, which even not print the "Success" info. wifi.sta.status() would change from 1 to 4.

Then I used the wifi.sta.config() to connect to ap, but failed with wifi.sta.status() = 4: STATION_CONNECT_FAIL

marcelstoer, 2016-04-10:
Feel free to re-open if you're still seeing this with the latest `dev` branch. Note that smartconfig is now turned OFF by default.

closed 2016-04-10 by marcelstoer