nodemcu/nodemcu-firmware #494: Filling & Fixing the document  [Here is a log]

MarsTechHAN, 2015-06-22:
<h3> If you find any bug inside the document, comment at here </h3> 
<h4>Now we are taking step fixing the</h4>
</br>
We will list them up here

- [ ] Add Cypto Modules
- [ ] Add RC Modules
- [ ] Add DHT Modules
- [ ] Add u8g to wiki(no only dev branch)
- [ ] Add 'hidden' function to wiki(like file.fsinfo()...those undoumented function)
- [ ] Fix the bug inside the existance Document

marcelstoer, 2015-06-22:
Once you're done you can close #486.

TerryE, 2015-06-22:
@MarsTechHAN, I am not quite sure what you mean by the title.  Are you asking for corrections to the user [API document](/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en) in the github wiki?  I am sure that a few of the community would be happy to support you here, but we just need to understand what you asking for.  Regards Terry

MarsTechHAN, 2015-06-22:
@TerryE Yes, nowaday, we really need to take out some time to updata the document after serval month's developing. Because the API document is really a mess... many powerful function are un-document, and many of the demo program are wrong. That will really consume new developer's a lot of time, which is what everybody dont want to see~

So...This page will work like this->
1.If some one comment a issue or unlist function in the NodeMCU, I will add them into the waiting list.
2.Everybody can go finishing the issue on the list, and after you have finished, comment that you have done some part of waiting list.

TerryE, 2015-06-22:
@MarsTechHAN, Some of the examples are really not good examples of how to program applications in Lua, and we have lots of overlaps eg.in fragment.lua with other examples.  A big job :worried:

MarsTechHAN, 2015-06-22:
@TerryE Yeah... That's the reason why we need to make a to-do-list

vowstar, 2015-06-26:
Yes, very big job.
so much examples need fix, we  just finished a little.

marcelstoer, 2015-12-12:
Documentation is now tracked in other issues.

closed 2015-12-12 by marcelstoer