nodemcu/nodemcu-firmware #497: Build should not depend on artifacts outside the repository

marcelstoer, 2015-06-24:
The build on Travis fails every now and then because the download of `https://raw.githubusercontent.com/GeorgeHahn/nodemcu-firmware/travis/tools/esp-open-sdk.tar.gz` is interrupted or aborted.

There's an example at https://travis-ci.org/nodemcu/nodemcu-firmware/builds/68076734 which was triggered from PR #496:

```
[0K$ wget https://github.com/GeorgeHahn/nodemcu-firmware/raw/travis/tools/esp-open-sdk.tar.gz -O tools/esp-open-sdk.tar.gz
--2015-06-23 22:43:18--  https://github.com/GeorgeHahn/nodemcu-firmware/raw/travis/tools/esp-open-sdk.tar.gz
Resolving github.com (github.com)... 192.30.252.128
Connecting to github.com (github.com)|192.30.252.128|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/GeorgeHahn/nodemcu-firmware/travis/tools/esp-open-sdk.tar.gz [following]
--2015-06-23 22:43:19--  https://raw.githubusercontent.com/GeorgeHahn/nodemcu-firmware/travis/tools/esp-open-sdk.tar.gz
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 23.235.46.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|23.235.46.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 35355361 (34M) [application/octet-stream]
Saving to: `tools/esp-open-sdk.tar.gz'

 0% [                                       ] 0           --.-K/s              
45% [================>                      ] 16,153,993  82.3M/s   in 0.2s    

2015-06-23 22:43:19 (82.3 MB/s) - Read error at byte 16153993/35355361 (Connection reset by peer).
```
As long as the firmware build depends on that file it should be incorporated into this repository.

MarsTechHAN, 2015-06-24:
But,I believe...
 All the things are on the Github, whatever who's repository; even the files is on NodeMCU, it still have its chance to fail...Beacuse the SDK is really big...

marcelstoer, 2015-06-24:
Correct, but running `git clone` is quite different from running `wget`. The point is to make sure that once Travis is done with `git clone` it has all it needs to build the firmware.

MarsTechHAN, 2015-06-24:
You idea might be a good choice, have you ever tried?

marcelstoer, 2015-06-24:
Yes, I added the SDK to `/tools` and removed the `wget` instruction from `.travis.yml` as seen in https://github.com/marcelstoer/nodemcu-firmware/commit/1b7d7dd509f86382776e4c01e24ed09337446ac3. The build on Travis worked like a charm.

```
```

MarsTechHAN, 2015-06-24:
If it works well on your repository, why dont you try to merge to dev branch and see how it goes?

PR #499 (2015-06-25): Keep SDK internally rather than fetching the one from GeorgHahn
Fixes #497.
commit 0ea12bdd (2015-06-25): Keep SDK internally rather than fetching the one from GeorgHahn, fixes #497
closed 2015-07-03 by vowstar