nodemcu/nodemcu-firmware #498: Firmware mess-up

brightmoon03, 2015-06-24:
When I  downloaded the flasher and firmware, and plugged in my node MCU through USB,  I hit the start button and it started to work but then it said that there is a connection timeout error. Now my Node MCU seems to have half of it's firmware on it and the blue LED isn't flashing on and off like it used to. Can  anyone help?

MarsTechHAN, 2015-06-24:
I believe you use the "FLASH_DOWNLOAD_TOOLS_v1.2_150512" which release by the Espressif which dont support Auto Download.

Try this out
```html
https://github.com/nodemcu/nodemcu-flasher
```

vowstar, 2015-06-24:
And needn't press any button.

brightmoon03, 2015-06-24:
When I plug in the Node MCU to both my Mac and my PC, my computers won't recognize it. And the blue led only flashes once, when you plug it in.

MarsTechHAN, 2015-06-24:
You may need to install the driver, the chip is called CP2102

brightmoon03, 2015-06-24:
I already installed the driver on my Mac. Do I need to install it on my PC? 

Thanks, Samuel Martin

saitejalakkimsetty, 2015-06-24:
@brightmoon03 I had similar issues before. I always endup reinstalling the driver, flash the espressif firmware on it and then install the nodemcu firmware. If I skip installing the espressif firmware, sometimes it works, sometimes it doesn't.

TerryE, 2015-12-12:
See #719.

closed 2015-12-12 by TerryE