nodemcu/nodemcu-firmware #514: Setting many gpio.modes consecutively causes chip to hang

liquiddandruff, 2015-06-28:
Came across this recently. Wonder what's the reason for this?

```lua
for i = 1, 10 do
    print("setting pin " .. i .. " as an OUTPUT")
    gpio.mode(i, gpio.OUTPUT)
    --tmr.delay(100000) --uncommenting this sets all pins from 1 to 9 without hang, hangs still on 10
end
```

MarsTechHAN, 2015-06-28:
9 and 10 only available on Amica Board, and if you set 9/10 on other ESP8266 modules(like NodeMCU Dev kit 0.9), it will make the flash unable to work.
[Beacuse in QIO mode, GPIO9/GPIO10  are data channel.And amica using DIO]

So which board you use? Can you take a picture of it?

liquiddandruff, 2015-06-28:
Ah yes, it actually hangs on both 9 and 10 for my esp201s and esp12es. Setting the pin modes in a loop from 1-8 is ok now.

closed 2015-06-28 by liquiddandruff

vowstar, 2015-06-29:
Don't touch the pin which UART uses.