nodemcu/nodemcu-firmware #515: 0.9.6-dev_20150625: GPIO2 cannot be used anymore, DHT11 code

ioddevs, 2015-06-28:
Something wrong with GPIO allocation, GPIO2 should have index 4.
It was working on 0.9.6-dev_20150406, yet cant run since 0.9.6-dev_20150625

Float binary used
on 0.9.6-dev_20150406 code is running fine, although crashed when timer loops
> tmr.alarm(2, 5000, 1, function() ReadDHT() end )
> checksum 	95
checksumTest 	95
Humidity:    63%
Temperature: 32 deg C
PANIC: unprotected error in call to Lua API (error loading module 'testdht' from file 'testdht.lua':
	not enough memory)

on 0.9.6-dev_20150625
checksum 	131
checksumTest 	129
Humidity:    0%
Temperature: 124.00390625 deg C

Below are code used following example from
http://scottsnowden.co.uk/esp8266-lua-dht11-temperature-and-humidity-sensor/
sensorType="dht11"          -- set sensor type dht11 or dht22
 
    PIN = 4 --  data pin, GPIO2
    humi=0
    temp=0
    --load DHT module for read sensor

vowstar, 2015-06-28:
Hello, 
We already have dhtlib and support all kinds of dht-sensor, including dht11, 21, 22, 33, 44 and after tested. This lib can detect sensor type automatically.
https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#dhtread,
Thank you very much for support NodeMCU opensource project.
Please try our latest release.
https://github.com/nodemcu/nodemcu-firmware/releases
```lua
pin = 5
status,temp,humi,temp_decimial,humi_decimial = dht.read(pin)
if( status == dht.OK ) then
  -- Integer firmware using this example
  print(     
    string.format(
      "DHT Temperature:%d.%03d;Humidity:%d.%03d\r\n",
      temp,
      temp_decimial,
      humi,
      humi_decimial
    )
  )
  -- Float firmware using this example
  print("DHT Temperature:"..temp..";".."Humidity:"..humi)
elseif( status == dht.ERROR_CHECKSUM ) then
  print( "DHT Checksum error." );
elseif( status == dht.ERROR_TIMEOUT ) then
  print( "DHT Time out." );
end
```

mtibi75, 2015-07-12:
Hi Iodevvs!

I have same problem, dht22 working fine with 20150311 and 0406 firmware, but not work 20150704 and 0625. the code and circuit not changed. any idea?

tibi

ioddevs, 2015-07-12:
hi tibi,
i havent relooked into the new lib vow mentioned.
i went back to another sample code, i forgot from where. 
when i have time i will check again.
thx

closed 2015-07-12 by ioddevs