nodemcu/nodemcu-firmware #516: CI and automated testing

marcelstoer, 2015-06-29:
I know core-contributors are very busy (SDK changes, docs, etc.). Still, I believe at some point this project should invest in CI and automated testing.

I'm eagerly waiting for the [WiPy MicroPython boards from Kickstarter](https://www.kickstarter.com/projects/wipy/the-wipy-the-internet-of-things-taken-to-the-next). They just recently described how they test.

> Rock solid software
> 
> We just finished building a professional integration test setup for the WiPy. This setup consists of a Linux box (a Raspberry Pi 2B in this case) running Jenkins and is connected to a WiPy via USB and also via WiFi. Every time a new software change is pushed to the MicroPython repository the following happens:
> 
> - The latest revision of the software is checked out and built.
> - The WiPy is updated via FTP with the recently built software.
> - The MicroPython test suite is executed on the WiPy.
> - In case of any failure, tests results are emailed to the developers.
> ... and we are busy right now writing specific tests for all the WiPy peripherals. Rock solid software

source: https://www.kickstarter.com/projects/wipy/the-wipy-the-internet-of-things-taken-to-the-next/posts/1277814

IMO that's the way to go (i.e. that's how I develop software in the non-embedded world).

MarsTechHAN, 2015-06-29:
I guess in the future, NodeMCU Dev Kit V2, we will enable some dramtic function (like modfible on-board automatic test system).
For now... I guess we can use a computer(like respe pie) with script to deal with this.
But how to co-wroking with Travis CI, do you have some idea?

marcelstoer, 2015-06-29:
> Travis CI is a wonderful system

Oh yes, truly marvelous!

> But how to co-wroking with Travis CI, do you have some idea?

My initial ideas....

The current build has one major flaw in my eyes: the binaries built by Travis are not use for any further processing, they just sit there until Travis dumps the VM it created to run the build.

So, the first step really would be to get the binaries off that Travis VM. In the `after_success` phase you could:
- `curl` them to a location of your choice
- `FTP` them to a location of your choice
- use the [Amazon S3 deployer](http://docs.travis-ci.com/user/deployment/s3/) provided by Travis
- use the wonderful [GitHub API](https://developer.github.com/v3/) to push the binaries back to your GitHub repo
Once this is done the "test controller" (e.g. Ra Py) needs to be notified somehow (e.g. [Travis webhooks](http://docs.travis-ci.com/user/notifications/#Webhook-notification)), download the binaries from where Travis put them, flash the ESP hooked up to the test controller, and run the tests.

borromeotlhs, 2015-07-21:
Why don't you all use docker to create a containerized linux image that has the xtensa toolchain in it by default?  You could then use dockerfiles to automate pulling the current firmware code from this repo, build it, and then have those prebuilts exported wherever you needed (as all contents of a docker container can be fetched from the 'host' docker machine).

You could then also put those very same 'push back to git repos' with Dockerhub's automated build system.  You would cut out all but the Travis CI parts that you truly want to focus on, plus, you'd be able to more quickly reproduce bugs.

marcelstoer, 2015-07-21:
> Why don't you all use docker to create a containerized linux image

I was working on PR #571 as you were writing this...

> have those prebuilts exported wherever you needed

Travis supports more than 25 providers out of the box: http://docs.travis-ci.com/user/deployment/ If nodemcu had an account with any of those it'd be trivial to tell Travis to deploy the binaries there.

> You could then also put those very same 'push back to git repos'

That's not a good idea. PR #560 was just recently merged.

marcelstoer, 2016-04-10:
@jmattsson do you have any input for this? IIRC you run something like this at $work?

jmattsson, 2016-04-11:
We do, for a great many things, just not NodeMCU at this stage. A typical setup comprises:

 - 2 (or more) devices to be tested. One sender, one receiver in the simplest case.
 - A Raspberry Pi to drive the test.
 - The devices are permanently hooked up to the Pi for both programming and control (in some cases the programming is via dedicated JTAG port, then control over UART).
 - Test scripts often written in Cucumber (or dialect thereof) with a small comms library for talking to a device over a UART and an enhanced version of the `expect` functionality (the default Ruby one has several shortcomings we discovered). When not using cucumber, we're usually using tcl/tk `expect` directly.
 - Devices get chip-erased and flashed for each test-run.
 - The whole test job is typically run as a Jenkins project with a junit formatter to capture test results prettily, and then feed into our LED strips of build-status lights.

A big challenge for this project would be to set up the actual hardware somewhere that Travis (or whichever CI is driving) can push both firmware and test script to, and get the results. Effectively it will allow running arbitrary code, so I think it needs to be a on a dedicated, partitioned network for anyone to feel even remotely comfortable hosting it.

karrots, 2016-09-08:
If only there were a way to emulate a ESP8266 chip.

https://github.com/espruino/Espruino/blob/master/targets/esp8266/emulator/README.md

borromeotlhs, 2016-09-10:
@jmattsson , have you all seen the post at http://tech.scargill.net/espressif-save-to-flash-an-alternative/#highlighter_574086 about faster flashing?  It may help get the testing done even faster if you can speed up flashes since you erase on retest, no?

closed 2019-06-21 by stale[bot]