nodemcu/nodemcu-firmware #517: RTC accuracy

patrickhwood, 2015-06-29:
I've run some experiments with the RTC (tmr.time()) and found that it's not very accurate when the device is active (most of my tests have been with the dht driver).  It drifts by up to 10 seconds per hour in some cases.

I've also found that the internal usec clock, system_get_time(), is far more accurate (at least on one test board; I will try others soon), presumably because it's connected directly to a divider on the xtal.  I replaced the code for tmr.time() to use this clock, compensating for when it rolls over every ~4000 sec, and found it drifted ~2 seconds over 10 hours, when compared against an NTP synchronized linux system.

Has anyone done any real tests of the various clocks available in the SDK?

marcelstoer, 2015-06-29:
Maybe PR #519 fixes this?

md5crypt, 2015-06-30:
ok, there is a problem with my code, value of system_rtc_clock_cali_proc is not constant over time. So the error does not accumulate but the time looks like this:
```
tmr.alarm(0,1000,1,function() print(tmr.time()) end)
570
577
573
574
576
```
I thinking of applying a low pass or median filter on it, but the question is how much memory can I waste on the filter? More memory -> more stable time

patrickhwood, 2015-06-30:
I would suggest using system_get_time() instead.  It is very stable (~2 sec over 10 hours on my board), and doesn't require normalization with system_rtc_clock_cali_proc.  It probably runs off the same xtal input as the RTC anyway.

md5crypt, 2015-06-30:
no, from what I read the RTC clock is a separate low-power circuit (probably a low-frequency RLC oscillator, that would explain the need of calibration) and it keeps on running even in deep sleep while the system time doesn't. I propose:
* tmr.systime() returns system_get_time()
* tmr.setrtcfactor(low,high) where low is a 31 bit fractional part and high 31 bit decimal part
* tmr.rtctime() returns rtc * factor

patrickhwood, 2015-06-30:
An RLC is probably going to drift with temperature changes, and maybe with voltage drift, depending on the quality of the regulator.  You're right about deep sleep mode, though, which is unfortunate.

Maybe by adding an optional byte string parameter to tmr.time(), it could be periodically synched from an NTP source's 64-bit timestamp.

saitejalakkimsetty, 2015-07-07:
@patrickhwood Where is the `system_get_time()` located and how do I use it?

md5crypt, 2015-07-07:
for an usage example see function tmr_now in tmr.c. It returns a counter that keeps on counting micro seconds (and overflows every ~1h). It isn't defined anywhere, only it's address is given in the link script. Generally to change from the rtc to system_get_time you only need to modify two lines of code (```uint32_t current = system_get_rtc_time()``` and ```lua_pushinteger(L, rtc2sec(rtc_time.block))``` to be specific ^^)

TerryE, 2015-11-02:
The new `dev`RTC modules address this issue.

closed 2015-11-02 by TerryE