nodemcu/nodemcu-firmware #518: RTC accuracy

patrickhwood, 2015-06-29:
I've run some experiments with the RTC (tmr.time()) and found that it's not very accurate when the device is active (most of my tests have been with the dht driver).  It drifts by up to 10 seconds per hour in some cases.

I've also found that the internal usec clock, system_get_time(), is far more accurate (at least on one test board; I will try others soon), presumably because it's connected directly to a divider on the xtal.  I replaced the code for tmr.time() to use this clock, compensating for when it rolls over every ~4000 sec, and found it drifted ~2 seconds over 10 hours, when compared against an NTP synchronized linux system.

Has anyone done any in-depth tests of the various clocks available in the SDK?

marcelstoer, 2015-06-29:
Isn't this a dup of #517?

patrickhwood, 2015-06-29:
Yeah, sorry.

closed 2015-06-29 by patrickhwood