nodemcu/nodemcu-firmware #520: HTTPS (SSL) issue

kroegerama, 2015-06-30:
The following code does not seem to work as expected ***(0.9.6-dev_20150627)***:
```
sk=net.createConnection(net.TCP, 1)
sk:on("receive", function(sck, c) print(c) end )
sk:on("connection", function(conn)  print("connected") end )
sk:on("disconnection", function(conn)  print("disconnected") end )
sk:connect(443,"api.pushbullet.com")
```
This instantly displays the "disconnected" message.
Same for:
```
sk:connect(443,"google.com")
```

pastukhov, 2015-07-01:
I'm getting 
```lua
DNS retry 1!
DNS retry 2!
DNS retry 3!
DNS retry 4!
DNS Fail!
````

MarsTechHAN, 2015-07-01:
If it just show disconnection...
It means NM even not connected...

Check the address

As for DNS Retry, the IP address is not exist in DNS Server

Both of question is not related much to SSL

kroegerama, 2015-07-01:
My question is indeed related to SSL. The error happens only, if I try to use a secure connection.

This code works just fine:

```
sk=net.createConnection(net.TCP, 0)
sk:on("receive", function(sck, c) print(c) end )
sk:on("connection", function(conn)  print("connected") end )
sk:on("disconnection", function(conn)  print("disconnected") end )
sk:connect(80,"api.pushbullet.com")
```

(changes: **0** in line 1, **80** in line 5)

@MarsTechHAN: Did you actually try connecting to a https service?

MarsTechHAN, 2015-07-01:
I will try it later, but does the server accept SSL connection？

kroegerama, 2015-07-01:
Thank you for trying to help.

Yes, the server does accept SSL connections. In fact, it requires SSL.
My non-ssl request just gets an answer from the Server like "please use SSL to connect to this service".

kroegerama, 2015-07-03:
Just a quick update: I built NodeMCU from the latest dev096-branch and still have the same issue.

marcelstoer, 2015-07-03:
I went as far back as NodeMCU 0.9.5 build 20150318 and can confirm this fails for SSL.

georgkreimer, 2015-07-09:
AFAIK dev096 branch is built on Espressif SDK 0.9.6_b1. It's using a 5month old libssl (https://github.com/nodemcu/nodemcu-firmware/commits/dev096/lib/libssl.a) SSL was properly fixed in a much later SDK version. I believe 1.1.0

abl, 2015-07-10:
My notes say that the last SDK with "working" (defined as: can connect to Parse) SSL prior to 1.1.0 (and even then a separate patch was required) was 0.9.3 - so 0.9.5 won't help.

dfeinzeig, 2015-10-20:
What is the latest with SSL support?  I've read and tried a bunch of things, including the above example, and SSL doesn't seem to work.

marcelstoer, 2015-10-20:
Did you try the `dev`/`dev140` branches which are based on the 1.4 SDK?

dfeinzeig, 2015-10-20:
I didn't realize it was ready for use.  I'm glad to try.  I can use a build of 140 from here - http://frightanic.com/nodemcu-custom-build/ ?

marcelstoer, 2015-10-20:
Yes.

drio, 2015-10-20:
Some more info related to this. Trying to run the nodemcu equivalent of this:

```sh
$ curl -v -u user@foo:pass  https://metrics-api.librato.com/v1/metrics
```

```
NodeMCU custom build by frightanic.com
        branch: dev140
        commit: 340ef8fc0ae00be1960acebe0276702dc7a00cb0
        SSL: true
        modules: node,file,gpio,wifi,net,tmr,uart
 build  built on: 2015-10-20 22:56
 powered by Lua 5.1.4 on SDK 1.4.0
```

```lua
wifi.setmode(wifi.STATION)
wifi.sta.config("is0xxx07","sxxxxxxxxsh")
wifi.sta.connect()

tmr.alarm(3, 5000, 0, function()
    print('ip:   ', wifi.sta.getip())
    local conn=net.createConnection(net.TCP, 1)
    conn:on("receive", function(conn, payload) print("payload: " .. payload) end)
    conn:connect(443, "metrics-api.librato.com")
    conn:send("GET /v1/metrics HTTP/1.1\r\n"
      .. "Authorization: Basic ZHJpbxxxxxYzZxxxTEwYjFkZGVmNA==\r\n"
      .. "Host: metrics-api.librato.com\r\n"
      .. "Connection: keep-alive\r\n"
      .. "Accept: */*\r\n\r\n")
end)
```

The `receive` callback never gets executed.

-drd

dfeinzeig, 2015-10-21:
What happens if you try google.com over https without the authorization header?

drio, 2015-10-21:
> What happens if you try google.com over https without the authorization header?

Same issue.

It seems work is being done in this area: https://github.com/esp8266/Arduino/issues/43

While reading the comments in that PR I thought about a possible alternative to help me 
with what I am trying to do. I just want to securely send small bits of data from my ESP to 
another server (internet server).

I was trying to directly connect to the server via https but an alternative may be connecting to a local server (maybe a raspberry pi) via regular http but encrypt the payload with RSA or some other asymmetric algorithm. Then the server would proxy the  request via https. I would still have to implement the algorithm but there seems to be a few implementations in lua (or C) out there.

How do you send secure data from your nodemcu devices?

dfeinzeig, 2015-10-21:
Interesting thread in the PR you reference.

I too just want some way to securely send data to an endpoint.  I was also thinking about keeping it inside my network and having something else communicate out, but that just feels like an awful solution for my purposes, because then I have two things to maintain.  :/

I'm curious how others are handling this?  Or is everyone just sending in the clear?

TerryE, 2015-10-21:
> I too just want some way to securely send data to an endpoint

Securely or privately or both? I want security but am not bothered about privacy so I sign my packets with an MD5 checksum which includes a shared secret.

drio, 2015-10-21:
Both.

drio, 2015-10-21:
> but that just feels like an awful solution for my purposes, because then I have two things to maintain. :/

Agreed,  that is the downside of this approach. 

It seems the SSL support is crucial. All the services use http over SSL. I was playing with Amazon IoT and they also only support https (and MQTT). And it seems they provide support for arduino. Not sure how that is done. I thought the ssl size was too much for arduino.

drio, 2015-10-22:
@TerryE Kudos on your unofficial [FAQ](http://www.esp8266.com/wiki/doku.php?id=nodemcu-unofficial-faq). Great resource.

drio, 2015-10-22:
Ok, I think this is what I am going to end up doing. We have hmac support at lua level (https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/crypto.c#L136) so I can verify the [integrity of the data](https://pymotw.com/2/hmac/) exchanged between my sensor and the server.

devsaurus, 2016-02-10:
@kroegerama @pastukhov @drio the code base saw several improvements since you reported these issues. Can you please recheck your testcases with current `dev`?

pastukhov, 2016-02-11:
Looks like the first code example is working.

devsaurus, 2016-02-20:
Thanks Artem! So we have a confirmed fix for the initial problem description.
I'm going to close this issue now. Everybody feel free to issue a new one in case other use cases are still showing erratic behaviour.

closed 2016-02-20 by devsaurus

seblanc, 2016-03-13:
Hi all,
I've been trying making an SSL request from google.com but with no success. I used the following code:

```LUA
conn=net.createConnection(net.TCP, 1)
conn:on("receive", function(sck, c) print(c) end )
conn:on("connection", function(conn)
      print("connected")
      conn:send("HEAD / HTTP/1.1\r\n".. 
             "Host: google.com\r\n"..
             "Accept: */*\r\n"..
             "User-Agent: Mozilla/4.0 (compatible; esp8266 Lua;)"..
             "\r\n\r\n") 
end )
conn:on("disconnection", function(conn)  print("disconnected") end )
conn:connect(443,"google.com")
```

Nothing is triggered (not even the "connected").

I also downloaded the *latest* version of nodemcu (master branch) from the website http://nodemcu-build.com by checking the SSL support.

```
NodeMCU custom build by frightanic.com
	branch: master
	commit: c8037568571edb5c568c2f8231e4f8ce0683b883
	SSL: true
```

Could someone tell me what I am doing wrong?

TerryE, 2016-03-13:
@seblanc sorry, but this is the wrong place to ask this type of question.  Please see #719 or our Support Page for help on application issues.

seblanc, 2016-03-14:
Ok. Thanks for the reply. As recommended, I posted my issue on StackOverflow:
http://stackoverflow.com/questions/35988328/https-ssl-get-request-with-nodemcu