nodemcu/nodemcu-firmware #521: official lua loader

Tom-Neverwinter, 2015-07-01:
the current methods of uploading lua code are not reliable. a official loader would be much easier. no more baud rate issues or outdated loader.

marcoskirsch, 2015-07-01:
I'd like to add to this request: an official, cross-platform, command line uploader. Not lua or even text only. Ideally written in python so it's easier to understand and maintain.

I've used luatool (https://github.com/marcoskirsch/luatool) and nodemcu-uploader (https://github.com/marcoskirsch/nodemcu-uploader). nodemcu-uploader allows you to upload binary files which is useful for uploading images on the nodemcu-httpserver project.

marcelstoer, 2015-07-01:
I don't know how "official" you'd like things to have. I use [esptool](https://github.com/themadinventor/esptool) for flashing and [ESPlorer](http://esp8266.ru/esplorer/) for everything else. Both are cross-platform (Python & Java) and work fine for me.

In https://github.com/nodemcu/nodemcu-flasher/issues/4#issuecomment-102686105 I suggested the NodeMCU folks who provide [nodemcu-flasher](https://github.com/nodemcu/nodemcu-flasher) should join forces with the esptool crew to bundle resources.

TerryE, 2015-07-01:
> Ideally written in python so it's easier to understand and maintain.

Why not in Lua?  Then we don't expect programmers to be familiar with Python *and* Lua.

MarsTechHAN, 2015-07-02:
@Terry @marcoskirsch We have a NodeMCU downloader which written in Python.

Tom-Neverwinter, 2015-07-02:
This is why I want an official loader; https://www.youtube.com/watch?v=idAo5IroCMw
The nodemcu firmware loader works, no problem, the other loaders just don't work.
if something is wrong with my setup its probably due to bad documentation, as the esp8266.com site has several of the same.

MarsTechHAN, 2015-07-02:
QAQ We are taking step fixing the document...As for offical one, yes. but in .NET which means not all-platform-support

Tom-Neverwinter, 2015-07-02:
@MarsTechHAN Thank you!!!

MacDada, 2015-07-12:
> As for offical one, yes. but in .NET which means not all-platform-support

That's too bad, @MarsTechHAN why would you switch from one platform–dependent solution (Flasher written in Delphi) to another platform–dependent solution (.NET)?

Does it mean that there will always be problems and hacks needed on Macs?

j3graham, 2016-01-04:
How about adding a TFTP server implementation to nodemcu? This would allow using any of the TFTP clients that exist for every imaginable language and platform. There are existing libraries for Python that could be used to do the heavy lifting in a nodemcu-specific tool. Writing a TFTP client in Lua would also be doable.

pjsg, 2016-01-04:
Please do not add a totally insecure protocol to nodemcu -- it will only create more problems than it solves.

What is the real requirement? Please say why the existing solutions do not meet your requirements.

I use the nodemcu-uploader.py (on windows no less!) and esplorer. I'd use esplorer only if it had a way to upload multiple files in a single go.

j3graham, 2016-01-04:
I wouldn't suggest having the TFTP server running at all times. It'd be a tool to be used when it was appropriate. Transferring files at development time would seem like such a situation.

My primary concern with the status-quo is the fragility inherent in tools using an interface intended to be used by a human rather than a well defined protocol for which there are mature implementations. The externally visible benefits would be improved reliability and performance.

This is likely missing the point of the original request, which I think was to bundle at least one simple tool with nodemcu itself that would be known to work with the matching version of the firmware. My suggestion pertained more to how this could be accomplished in a generic way that could provide benefit for other tools and other use cases.

TerryE, 2016-01-04:
We have no plans to implement TFTP.

TerryE, 2016-01-04:
Sorry my intent was to close of the TFTP thread.  Having an "official" loader is not quite closed, though we might introduce one under #772

renevdzee, 2016-01-22:
For those interested in uploading files using tftp, I wrote a small tftp server in lua for put requests. See https://github.com/renevdzee/nodemcu_tftpd

marcelstoer, 2016-04-10:
> Having an "official" loader is not quite closed

Terry, I don't think this is going to happen in the foreseeable future. Even though you spend a lot of time for this and we've got more committers than 12 months ago it's just about enough to keep the firmware going.

- tools are documented at http://nodemcu.readthedocs.org/en/dev/en/upload/
- #1215 would introduce auto-baud support
- ESPlorer is cross platform and is very popular AFAIK

-> I'll close this

closed 2016-04-10 by marcelstoer