nodemcu/nodemcu-firmware #524: wifi.sta.status() constants missing

zerog2k, 2015-07-03:
wifi.sta.status() returns number： 0~5
    0: STATION_IDLE,
    1: STATION_CONNECTING,
    2: STATION_WRONG_PASSWORD,
    3: STATION_NO_AP_FOUND,
    4: STATION_CONNECT_FAIL,
    5: STATION_GOT_IP.

however, these do not seem to be constants, and probably should, e.g., wifi.STATION_GOT_IP

zerog2k, 2015-07-03:
looks like some attempt was made but commented out?
https://github.com/nodemcu/nodemcu-firmware/blob/dceed526c90aaa63a4ec2aaf11d8ce313f40c6ff/app/modules/wifi.c#L1101-L1106

dnc40085, 2015-07-14:
I my most recent PR(#551), when I added the event monitor I uncommented those constants so they could be used with the new functions.

The constants are:
- wifi.STA_IDLE
- wifi.STA_CONNECTING
- wifi.STA_WRONGPWD
- wifi.STA_APNOTFOUND
- wifi.STA_FAIL
- wifi.STA_GOTIP

zerog2k, 2015-07-18:
thanks, will look for those

closed 2015-07-18 by zerog2k