nodemcu/nodemcu-firmware #526: attempt to index global 'mqtt' (a nil value)

doringaiu, 2015-07-03:
##MQTT module issues

When i try to use the **mqtt module** and **mqtt.client sub-module** i always get this error when this line of code is executed:
```
m = mqtt.Client("clientid", 120, "user", "password")
```

The provided example named **Connect to MQTT Broker** or other files from lua examples folder like **mqtt_file.lua** don't work at all. Sometimes i get errors like *malformed number near xxx* ... 

I have the ***0.9.5*** firmware installed.

Aeprox, 2015-07-05:
The MQTT support has been greatly improved by many contributors and the devteam since the 0.9.5 build. Have you tried upgrading?

Also; the "malformed number.." error makes me guess it might have something to do with integer or float version of the firmware. Which are you using?

doringaiu, 2015-07-06:
Problem sovled by updating to 0.9.6

closed 2015-07-06 by doringaiu