nodemcu/nodemcu-firmware #527: callback functions mqtt

gprandst, 2015-07-03:
1.
When doing some subscribes consecutive, callback is always done from the last subscribe.

    m:subscribe("test/test_1",0,function con() print("Subscribe 1") end)
    m:subscribe("test/test_2",0,function con() print("Subscribe 2") end)
    m:subscribe("test/test_3",0,function con() print("Subscribe 3") end)   
Output:
  Subscribe 3
  Subscribe 3
  Subscribe 3

The same if consecutive publish's.

2.
Text at API mqtt.client:subscribe() seems to be wrong:

Parameters
topic: a string topic to subscribe to
qos: qos subscription level, default 0
function(client, topic, message): callback fired when message received.

Parameters for callback function doesn't exist.

benrussell, 2015-07-07:
Looks like C internals only support only callback function per client instance.

Data structure created here:
https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/mqtt.c#L1038

Callback function stored here:
https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/mqtt.c#L1114

Callback restored and called here:
https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/mqtt.c#L277

Also appears that the instance of m is pushed as an argument to your function.

Example code here appears to be correct:
http://www.nodemcu.com/docs/mqtt-module/

API reference here appears to be wrong:
http://www.nodemcu.com/docs/mqtt-client-module/

marcelstoer, 2016-01-17:
Pity this wasn't fixed with #774. So, it's still wrong at http://nodemcu.readthedocs.org/en/dev/en/modules/mqtt/#mqttclientsubscribe

devsaurus, 2016-01-30:
ad 1.
You need to nest the calls to `subscribe()`. Firmware will execute the callback function that happens to be active at the time where subscription succeeds. This can be any of the three functions in your example.

ad 2.
API doc at http://nodemcu.readthedocs.org/en/dev/en/modules/mqtt/#mqttclientsubscribe is updated now.

commit 9741c8f1 (2016-01-29): fix #527
closed 2016-01-30 by devsaurus