nodemcu/nodemcu-firmware #528: firmware file

dcjw, 2015-07-03:
Hi 
I am a newbie with esp8266 module, so I am sorry if I am asking dummy questions... there are a lot of files in this repository. Which one is the firmware? Do we have to build it?

marcelstoer, 2015-07-03:
Welcome to the esp8266  world. This issue list is not a support forum, use http://www.esp8266.com/viewforum.php?f=17 for that. Other helpful resources:
- http://www.esp8266.com/wiki/doku.php
- http://nodemcu.com/index_en.html

> Which one is the firmware? Do we have to build it?

- official releases here on GitHub https://github.com/nodemcu/nodemcu-firmware/releases
- http://frightanic.com/nodemcu-custom-build/ to have your own firmware built in the cloud (disclaimer: my service)

dcjw, 2015-07-03:
My bad, I think I found it, and thanks for showing me the rope! The support forum is really a rich database with lots of info & projects. Thanks a lot!

TerryE, 2015-07-03:
@dcjw, please do us a favour and close this issue so the nodeMCU guys don't have to :smile:

dcjw, 2015-07-03:
oh ok ;)

closed 2015-07-03 by dcjw