nodemcu/nodemcu-firmware #530: Add a function to show the nodemcu build info.

jiegec, 2015-07-04:
Something like vim --version, including the current nodemcu version, whether the modules are enabled.. built on which time... And add a rule to contributing by issues: put all these information first. In this way, we can easier to know whether the code has fixed in newer dev versions.

MarsTechHAN, 2015-07-04:
You can create a PR... 
Simply add a function and add #define on optional file.

jiegec, 2015-07-04:
Ok, but i am not at home until the tomorrow... And i may not be able to touch my pc because of the final exam -_-

MarsTechHAN, 2015-07-04:
@jiegec You can program with you phone, at least it is a Unix BSD Device(If you use iOS) or Linux (Amdroid)

marcelstoer, 2015-07-04:
>  ...including the current nodemcu version, whether the modules are enabled.. built on which time

This is related to https://github.com/nodemcu/nodemcu-firmware/issues/449#issuecomment-109283562 in which I documented how my public cloud build handles this.

jiegec, 2015-07-04:
@marcelstoer Yes, that's the same, just need a small PR to include a small function in it.

closed 2015-07-04 by jiegec