nodemcu/nodemcu-firmware #532: CI build for dev112 fails

marcelstoer, 2015-07-04:
As per https://travis-ci.org/nodemcu/nodemcu-firmware/builds/69487853#L2112 the `dev112` branch is currently broken.

dnc40085, 2015-07-04:
I compiled and burned dev112 and every time I reset the chip I get non-ascii characters. no info on the chip, no nothing, just non-ascii characters

felixcameron, 2015-07-05:
I did the same, the output is ascii but at 115,000bps I think

dnc40085, 2015-07-05:
I tried it with the default baud rate (9600) and when I tried connecting to it at 9600, 74880, 115200 baud rates I got nothing. I also tried changing the default rate to 115200 and got the same results.

marcelstoer, 2015-07-05:
Guys I appreciate your feedback and tests but the subject of this issue is the failing CI build. Even if your local builds at least compile it's important for me that the reference build on Travis is green.

vowstar, 2015-07-06:
We will try to pass the build, need more testing...

georgkreimer, 2015-07-08:
This is what I get on an ESP-01 at 115200
rl��r��#�n������p�<����8��ǒ��lp�nn��;�nĒ���b�$  rrp�n��܀��l���b�n��n�䎀l�p��nn����l`��#�n�$��l`�8`rn|��n����l`�~����#�n�$���nn���l`nn��܀������b�n��n�$��������8��nn���l`

pastukhov, 2015-07-09:
Same on ESP-201

marcelstoer, 2015-07-13:
I just updated the issue title because the exact same issue exists in the new dev120 branch. See https://travis-ci.org/nodemcu/nodemcu-firmware/builds/70753912 I hope you find the time to look into this soon.

vowstar, 2015-07-14:
The problem is in lwip, sdk 1.x broken the wifi due to unkown problem with open-source lwip, when we use espressif's closed source lwip, wifi works. Still finding solutions. Now CI may works.

georgkreimer, 2015-07-16:
@vowstar Any easy way to switch to the espressif lwip?

marcelstoer, 2015-07-16:
https://travis-ci.org/nodemcu/nodemcu-firmware/builds/70897645 is proof that the CI build is fixed.

closed 2015-07-16 by marcelstoer

georgkreimer, 2015-07-17:
@marcelstoer Have you gotten the latest build to run on an actual ESP8266?

marcelstoer, 2015-07-17:
Haven't tried. That isn't relevant for this issue though as it's about the CI build failing.