nodemcu/nodemcu-firmware #534: dev: ws2812 O2 optimize turns first led green

makefu, 2015-07-06:
with the addition of `__attribute__((optimize("O2")))` the ws2812 library does not work reliably anymore, the first LED is always green for me.

https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/ws2812.c#L19

I tried building the firmware without the attribute flag and the behavior is correct again

creationix, 2015-07-06:
Duplicate of #487.  Thanks for helping find the cause.

MarsTechHAN, 2015-07-07:
@makefu 
@creationix 
Maybe the version cause the problem. Can you guys show your NodeMCU's Version
(Like clone from which version which branch and using which kind of board)

makefu, 2015-07-07:
I tried most current 'dev' branch (as of 2015-07-05)  with the new `__attribute__((optimize("O2")))` flag in the `ws2812.c` file.
This creates the issue.
When i remove the attribute tag from the file the problem goes away.

DR-AU, 2015-07-09:
I use the pre-built firmware and from NodeMCU 0.9.6 build 20150625 the problem exists

TerryE, 2015-08-24:
Closed as duplicate of #487.

closed 2015-08-24 by TerryE