nodemcu/nodemcu-firmware #535: ow broken skip/select

jorgempy, 2015-07-06:
doing some testing using a ds2413 i discovered an issue with the ow module.
the LSB of the command sent after a skip or select is sent wrong, it always send '0'.
this cause issues in odd numbers (they lose a bit), but not in even numbers.
this is important because it explains why the people using ds18b20 sensors have no issues.
the rom commands used in these sensors are 0x44 and 0xBE (both even!!).
I conducted a little experiment using a logic analizer, using this code:
```lua
for i= 1,255,1 do
    ow.reset(pin)
    ow.skip(pin)
    ow.write(pin, i,1)
end
```
the issue can be reproduced, i attach a log of the analyzer, there you can see that the commands go in increments of two. digging through the ow issues i found somebody that replaced the ``` ow.skip(pin)``` for a ``` ow.write(pin, 0xCC, 1) ``` .
that fixed my problems, but for the people trying to select a device and then send a rom command with an odd number it can get a little tedious.
I found in onewire.c the answer: ``` ow.skip(pin) ==  onewire_write(pin, 0xCC, 0); ``` and the select is the same thing.
In the week i will prepare a pull request, in the meantime i leave this here in case someone cares to discuss if i should add a ``` power ``` parameter to skip/select or make the skip/select assume a '1' for power.

the (partial) log:
```
[...]
27,0.071034000000000,Reset Pulse
27,0.071556500000000,Presence Pulse
27,0.072168500000000,Skip Rom Command, 0xCC
27,0.072735500000000,Data, 0x1C
28,0.073659500000000,Reset Pulse
28,0.074182000000000,Presence Pulse
28,0.074794000000000,Skip Rom Command, 0xCC
28,0.075361000000000,Data, 0x1C
29,0.076280000000000,Reset Pulse
29,0.076802500000000,Presence Pulse
29,0.077414500000000,Skip Rom Command, 0xCC
29,0.077981500000000,Data, 0x1E
30,0.078900500000000,Reset Pulse
30,0.079423000000000,Presence Pulse
30,0.080035000000000,Skip Rom Command, 0xCC
30,0.080602000000000,Data, 0x1E
31,0.081516500000000,Reset Pulse
31,0.082039000000000,Presence Pulse
31,0.082651000000000,Skip Rom Command, 0xCC
31,0.083218000000000,Data, 0x20
32,0.084151500000000,Reset Pulse
32,0.084674000000000,Presence Pulse
32,0.085286000000000,Skip Rom Command, 0xCC
32,0.085853000000000,Data, 0x20
33,0.086781500000000,Reset Pulse
33,0.087304000000000,Presence Pulse
33,0.087916000000000,Skip Rom Command, 0xCC
33,0.088483000000000,Data, 0x22
34,0.089412000000000,Reset Pulse
34,0.089934500000000,Presence Pulse
34,0.090546500000000,Skip Rom Command, 0xCC
34,0.091113500000000,Data, 0x22
35,0.092037000000000,Reset Pulse
35,0.092560000000000,Presence Pulse
35,0.093172000000000,Skip Rom Command, 0xCC
35,0.093739000000000,Data, 0x24
36,0.094667500000000,Reset Pulse
36,0.095190000000000,Presence Pulse
36,0.095802000000000,Skip Rom Command, 0xCC
36,0.096369000000000,Data, 0x24
37,0.097293000000000,Reset Pulse
37,0.097815500000000,Presence Pulse
37,0.098427500000000,Skip Rom Command, 0xCC
37,0.098994500000000,Data, 0x26
38,0.099918000000000,Reset Pulse
38,0.100441000000000,Presence Pulse
38,0.101053000000000,Skip Rom Command, 0xCC
38,0.101620000000000,Data, 0x26
39,0.102539000000000,Reset Pulse
39,0.103061500000000,Presence Pulse
39,0.103673500000000,Skip Rom Command, 0xCC
39,0.104240500000000,Data, 0x28
40,0.105169000000000,Reset Pulse
40,0.105691500000000,Presence Pulse
40,0.106303500000000,Skip Rom Command, 0xCC
40,0.106870500000000,Data, 0x28
41,0.107794500000000,Reset Pulse
41,0.108317000000000,Presence Pulse
41,0.108929000000000,Skip Rom Command, 0xCC
41,0.109496000000000,Data, 0x2A
42,0.110420000000000,Reset Pulse
42,0.110942500000000,Presence Pulse
42,0.111554500000000,Skip Rom Command, 0xCC
42,0.112121500000000,Data, 0x2A
43,0.113040500000000,Reset Pulse
43,0.113563000000000,Presence Pulse
43,0.114175000000000,Skip Rom Command, 0xCC
43,0.114742000000000,Data, 0x2C
44,0.115666000000000,Reset Pulse
44,0.116188500000000,Presence Pulse
44,0.116800500000000,Skip Rom Command, 0xCC
44,0.117367500000000,Data, 0x2C
45,0.118286500000000,Reset Pulse
45,0.118809000000000,Presence Pulse
45,0.119421000000000,Skip Rom Command, 0xCC
45,0.119988000000000,Data, 0x2E
46,0.120907000000000,Reset Pulse
46,0.121429500000000,Presence Pulse
46,0.122041500000000,Skip Rom Command, 0xCC
46,0.122608500000000,Data, 0x2E
47,0.123523000000000,Reset Pulse
47,0.124045500000000,Presence Pulse
47,0.124657500000000,Skip Rom Command, 0xCC
47,0.125224500000000,Data, 0x30
48,0.126153000000000,Reset Pulse
48,0.126675500000000,Presence Pulse
48,0.127287500000000,Skip Rom Command, 0xCC
48,0.127854500000000,Data, 0x30
[...]
```

jas39, 2015-08-19:
skip, search and write_bytes are affected. The following workaround is ok:
present=ow.reset(pin)
 --ow.select(pin, addr) -- This command is buggy
 ow.write(pin,0x55, 1)  -- 0x55==search aka select command
 for i=1,8 do
	ow.write(pin,addr:byte(i), 1)
end

jorgempy, 2015-08-21:
Fixed by #608 thanks @jmattson

closed 2015-08-21 by jorgempy