nodemcu/nodemcu-firmware #536: Issue with communication between two esp's using arduino uno

thrinasco, 2015-07-07:
Hi,

This is venkat,Please ,I would request you to clear one of my doubt.i was using the following library for communication  
https://github.com/INEXTH/Arduino-ESP8266_libs

Server is created and when i am starting server to make communication between them iam getting the following error which was marked.

AT+CIPSTART=(4)("TCP"),("192.16ø.4.1"),(80)
Link typ ERROR

I am doing these commuinication using two esp8266 with two arduino boards.

​
[image]

thrinasco, 2015-07-07:
help me out

benrussell, 2015-07-07:
User has filed duplicate bug report here, where it belongs:

https://github.com/INEXTH/Arduino-ESP8266_libs/issues/3

Suggest closing this issue as not relevant to nodemcu project.

MarsTechHAN, 2015-07-07:
Also... Here is NodeMCU's Issue List. Why not try NodeMCU : )

devsaurus, 2015-12-13:
See #719.

closed 2015-12-13 by devsaurus