nodemcu/nodemcu-firmware #537: Blocking GPIOs for couple of micro seconds on Startup (To resolve Back EMF from load/Relays)

npadhya, 2015-07-07:
If the ESP is connected to a Relay with any of the GPIO, and is both the systems (ESP + Relay module)  are connected to the same power source (Basically turning both ON together ) than ESP does not boot up and could not connect to Wifi. 

If the GPIO is disconnected on startup and connected after could of seconds, that it works fine.

dnc40085, 2015-07-09:
I don't believe this is a bug or issue with the nodemcu firmware, sounds more like a hardware issue.
Do you have a freewheeling diode across the relay?
Do you have decoupling capacitors next to the esp8266 to filter any noise caused by other components in your circuit?

npadhya, 2015-07-09:
Freewheeling did the fix.

closed 2015-07-09 by npadhya