nodemcu/nodemcu-firmware #539: mqtt.client:connect() secure options

doringaiu, 2015-07-08:
##mqtt.client sub-module issues
###mqtt:connect( host, port, secure, function(client) )

There is no information about the security standard used here. Is it SSL 3.0 or TLS 1.1/1.2 ?

If so please provide some examples where the secure flag is 1 and more details about how to use TLS or at least SSL.

mangelajo, 2015-09-19:
+1, any info on this?

sulicz, 2015-09-28:
+1, any info?

grahamehorner, 2015-09-29:
+1 more info the better :D

TerryE, 2016-01-02:
I would _love_ of the guys who +1 this to do the research, retro-engineer the code, and raise a PR or comment to resolve this.  The committers here don't have the time, given their other priorities. 

Anyone can say +1. It's a lot harder to step up and actually do the work.

marcelstoer, 2016-01-05:
Terry, I share your frustration but I still disagree.

The way we spend our (free)time on this project should also be driven by what the users need. How can we ensure that what we spend time on is *relevant for the community* if there's no way for the community to tell us what *their* priorities are? Of course we're still free to set our own priorities, it's our time after all, but I'm glad we get those +1-style comments. Pity GitHub offers no (up/down) votes for issues.

I stumbled over this issue a few months ago and naively thought it should be pretty easy to answer. However, if you know my background it's no surprise that when I found https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/mqtt.c#L874 it was all Greek to me. I gave up :disappointed:.

tuxedo0801, 2016-01-05:
Had a look at the code:

https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/mqtt.c#L959

Question is: Is client-ssl "built in"? Or ist it disabled in build?

The method which get called if secure is build in and enabled ate something like "esconn_secure_*".

Looks like they should be defined in espconn.h and imlpemented in espconn.c:

https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/lwip/app/espconn.c

But there is no "secure" used in such files...

Seems to be prepared by mqtt code, but not implemented at all?

Still frustrated....

marcelstoer, 2016-01-05:
`#CLIENT_SSL_ENABLE` is defined in https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/include/user_config.h#L48. You define this when you build the firmware (e.g. on http://nodemcu-build.com/).

nickandrew, 2016-01-09:
The `espconn_secure_*` functions are in libssl.a to which I assume source code is not available.

The question can be answered by Espressif.

It should also be possible to test the ssl protocol by setting up a dummy server to log the connection from the client. How to do it:

```
$ openssl req -x509 -new -out test.cert                      
Generating a 2048 bit RSA private key
..............................+++
.........+++
writing new private key to 'privkey.pem'
Enter PEM pass phrase:
Verifying - Enter PEM pass phrase:
-----
You are about to be asked to enter information that will be incorporated
into your certificate request.
What you are about to enter is what is called a Distinguished Name or a DN.
There are quite a few fields but you can leave some blank
For some fields there will be a default value,
If you enter '.', the field will be left blank.
-----
Country Name (2 letter code) [AU]:
State or Province Name (full name) [Some-State]:
Locality Name (eg, city) []:
Organization Name (eg, company) [Internet Widgits Pty Ltd]:
Organizational Unit Name (eg, section) []:
Common Name (e.g. server FQDN or YOUR name) []:testhost.example.com
Email Address []:

$ openssl s_server -accept 12345 -debug -cert test.cert -key privkey.pem 
Enter pass phrase for privkey.pem:
Using default temp DH parameters
Using default temp ECDH parameters
```

An ssl server is now listening on port 12345. Just direct the NodeMCU mqtt.client to connect to it.

jmattsson, 2016-01-10:
According to a test report I saw retweeted by Espressif some time ago, it's TLS 1.2. [Here](https://twitter.com/TimMattison/status/672434186289049600) is the original tweet.

marcelstoer, 2016-01-10:
Thanks Johny, now documented at https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#mqttclientconnect.

closed 2016-01-10 by marcelstoer

trahloff, 2016-02-01:
Some tests show that the NodeMCU uses TLS 1.1 and not TLS 1.2: [https://github.com/nodemcu/nodemcu-firmware/issues/996](https://github.com/nodemcu/nodemcu-firmware/issues/996)

stgoddv, 2017-10-12:
Hi!
Any updates about this with the current firmwares?
Thanks!