nodemcu/nodemcu-firmware #540: expose ssl aes

sauloal, 2015-07-08:
Would it be possible to expose:
app/ssl/crypto/ssl_aes.c

so that we can use public key encryption to send data under insecure connections?

regards

peon374, 2015-10-13:
I would like to do encrypted auth to cloud service and this would help me too.

marcelstoer, 2015-10-14:
It's IMHO a bit unfortunate AES isn't part of the [crypto module](https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#crypto-module) but buried in the SSL stuff.

dfeinzeig, 2015-10-21:
I thought I saw mention that the esp8266 has optimizations for aes?

TerryE, 2015-11-02:
@dfeinzeig, yes it does, but the API is not documented.  If you wish to retro-engineer it and develop the enhancement then we would welcome this, but in the meantime those actually developing the core firmware need to set their own priorities.

devsaurus, 2016-02-20:
See related PR against `dev`.

closed 2016-02-20 by devsaurus