nodemcu/nodemcu-firmware #541: {SOFT,STATION}AP mode not working after reset on current branches

jmattsson, 2015-07-09:
I'm seeing a highly reproducible issue whereby a configured AP is brought up fine the first time, but after a reset the AP is no longer active and does not show up in scans any more and can't be connected to. Issuing the same commands that originally set up the AP does *not* result in it working, but configuring a *different* SSID successfully brings up an AP with that SSID, until the next reset.

This behaviour was introduced some time after the 0.9.5_20150318 release. I haven't managed to track it down to a precise commit. Around 0b01f28c62acc154db7f365d4e490252859b5906 it appeared that the problem wasn't as reproducible, and my git bisect failed. Before that commit I've seen it both work and fail, though during my git bisect commits later than that always failed. The commit in question does not seem relevant to this behaviour. At some points the failure mode was that it brought up an open network with the default SSID (ESP_<mactail>) instead of the configure AP!

I've tested the following branches/tags, with results as follows:
0.9.5_20150318:  works
master: fails
dev: fails
dev096: fails
dev112: fails, it's not even possible to connect to the AP initially

To reproduce:

    wifi.setmode(wifi.SOFTAP)
    wifi.ap.config({ssid='mynet',pwd='somesecret'})
      <confirm that 'mynet' shows up in a scan>
      <hit reset>
      <note that 'mynet' no longer shows up> 
    wifi.ap.config({ssid='mynet',pwd='somesecret'})
      <note that 'mynet' still doesn't show up>
    wifi.ap.config({ssid='changeme',pwd='somesecret'})
      <confirm that 'changeme' shows up in a scan>

Tested on a NodeMCU Dev Kit 0.9, mostly using STATIONAP, but same behaviour also observed in SOFTAP mode.

A very ugly workaround is to always start an AP on a temporary SSID on boot, and tmr.alarm() a reconfig onto the desired AP a little while later. Changing the SSID too quickly back-to-back does not work, there must be a bit of time in between.

dnc40085, 2015-07-09:
I have also experienced this issue.
My workaround is to switch to Station mode before reset using the following:
`wifi.setmode(wifi.STATION)
node.dsleep(10)`

saitejalakkimsetty, 2015-07-09:
@jmattsson This was the problem I was mentioning in #511

crazycoder1999, 2015-07-10:
hi, I confirm .. to have the same problem on the lastest firmware.

saitejalakkimsetty, 2015-07-10:
@crazycoder1999 do you see `mynet` after reboot?

crazycoder1999, 2015-07-10:
I succesfully connected to the wifi only the first time and succesfully did an "http get" to a server.
Since than I was not able to connect again to the AP... it stays in "connecting" state .
I'm able to see "mynet" scanning for AP.
I have rebooted the device a lot of time, but nothing is changed.
I will try later to connect to another AP and than connect again to "mynet".

jmattsson, 2015-07-14:
We have tracked this down to an issue in libnet80211.a (binary part of the SDK).

If the system is busy at the wrong time, the MAC Beacon timer doesn't get serviced in time, and as a result it dies. Once the beacons stop being transmitted, the network is effectively dead, even though the node still thinks it's alive. This is why it comes back to life if you *change* something about the AP - then the whole thing gets restarted and the beacon timer is properly reactivated.

This being effectively a timing issue also explains why I didn't manage to pinpoint a particular commit that introduced this problem.

I've tried to reproduce the issue on an SDK 1.1.2 project (I can't test using NodeMCU because the dev112 branch doesn't have working WiFi at the moment). As far as I can tell, the problem has been fixed in that version of the SDK. The worst I can manage to do to it is to seriously delay a beacon, but it properly resumes the correct timing right after.

So, I think we'll all just need to sit tight until @vowstar finishes his work on getting NodeMCU to run with SDK 1.1.2 (or newer ;)

TerryE, 2015-07-14:
@jmattsson, Jonny do you have a handle on the time sensitivity here.  I know that the Espressif folks don't document it well but their is an advisory added to the V1.2 (IIRC) SDK refman to the effect that *if any tasks run for more than 10 mSec, then services such as Wifi can fail*.  In the Lua case this is the running time of any Lua callback function.  I regularly see code examples which break this rule.

jmattsson, 2015-07-14:
@TerryE I don't have a definite measure for the NodeMCU / 0.9.5 side; It dies early in the boot process as far as I can tell. It doesn't seem that the beacon timer handler needs to be late by much at all before that causes it to fall over. I suspect a delta goes negative and then gets multiplied and/or sign-extended into oblivion. Probably a simple two-line fix if we had source... *mutters about !^@#!$ binary blobs*

When I tested with SDK 1.1.2 using esphttpd, I could sit in a blocking delay() for 800+ms without it killing the beacon timer. Of course, in the meantime no beacons were going out, but everything recovered once the delay exited.

TerryE, 2015-07-14:
> *mutters about !^@#!$ binary blobs*
Snap.  I really don't see what Espressif has to gain by doing this, unless of course doing so would expose some copyright or disclosure issues of their own.  Their IPR value is in the hardware not the SDK.  Fixing bugs would be a lot easier if we had access to the source.

jmattsson, 2015-10-28:
This problem should be gone now after the 1.4.0 upgrade on the dev branch. If you're still seeing it, please let me know.

pepsifan92, 2015-11-05:
Hi. The "problem" of loosing the connection to the AP (this issue, not the referenced #566 ?) should be gone in dev 1.4.0?
I got the problem with the very unstable SOFTAP (on NodeMCU 0.9.6 build 20150704) and can't find any help elsewhere. I'm confused, that the SOFTAP mode exists, but seems to be that unreliable that it's useless till now. 
Do I have to wait for the next release to use the SOFTAP mode?!

TerryE, 2015-11-06:
> Do I have to wait for the next release to use the SOFTAP mode?!
See #719 and try upgrading to a ev`build using the fritanic cloud service.

Retroaktiv, 2016-01-21:
Hello i have the Problem with my Iphone not logged in Soft AP.
the Dev Nodmcu (Fritanic) not Works. (http://nodemcu-build.com/)
MY solution on 0.9.6-dev_20150704: 
Setting up with init.lua two Wifi Softap!
 1.: Serverdummy.lua an then Server.lua.  The first is  the Dummy
and Number 2 is the real Server and works login.
Sorry for my bad endlish ! :-((

jmattsson, 2016-02-03:
Fixed by the SDK upgrade, closing.

closed 2016-02-03 by jmattsson