nodemcu/nodemcu-firmware #543: Achilles and the tortoise, 1.2 SDK

pastukhov, 2015-07-09:
What about 1.2 SDK version?

MarsTechHAN, 2015-07-11:
You can have a try : )

fabianhu, 2015-07-11:
Did a short test drive (compile the dev). 
Ended up on a tree.
Could not get communication with device over serial port.

MarsTechHAN, 2015-07-11:
@fabianhu  You tried the SDK 1.2? Can you explain detail phenomenon?

fabianhu, 2015-07-11:
Just exchanged the link to the SDK within my already installed toolchain. 
After downloading the compiled nodemcu, I could not establish connection with ESPlorer.
I tried several re-compiles and used [https://github.com/pfalcon/esp-open-sdk] newest version to install toolchain again, it brings SDK 1.2.0 with it.
The module sends a 
```
ets_main.c 
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)
```
with 74880 baud on reboot.
No communication with ESPlorer - tried with all baudrates from dropdown. (GPIO 0 floating!)
Sorry, I do not have more findings yet.

TerryE, 2015-12-12:
We are now on 1.4

closed 2015-12-12 by TerryE