nodemcu/nodemcu-firmware #546: New GPIO Method.

roccomuso, 2015-07-10:
Couldn't be useful add in the GPIO Module the method: gpio.toggle()
To switch the state of a GPIO pin?

Great job ;)

creationix, 2015-07-10:
I'm not sure that would be useful enough to be a dedicated API.  You do know that you can read output pins right (I didn't till I read the bitlash examples)?  Simply write the `not` of the read to toggle.

roccomuso, 2015-07-10:
Yeah i know it of course. It's just for completeness, read, write, toggle. Classic operations for a general purpose pin.

MarsTechHAN, 2015-07-11:
I believe this is useless... Why dont you just use gpio.mode?

nodemcu, 2015-07-30:
gpio.mode changes like direction but toggle changes state, or pin level.
but in sdk level there is no toggle function , so to do this have to read and write, this change pin mode from output to input and output again, without user awareness, maybe cause some problem? glitch
?

eyaleb, 2015-07-31:
Are you sure that reading an output pin changes the mode? I thought it just reads the register which, for output pins, holds the last state set.

nodemcu, 2015-07-31:
opps, this is old setup, it will not put into input state by nodemcu firmware according to this commit #df3c82391ffbe71223ddc6cd6f1d3e38989c7b3b
https://github.com/nodemcu/nodemcu-firmware/commit/df3c82391ffbe71223ddc6cd6f1d3e38989c7b3b#diff-2eeba06bdd37586901dca4f122de1cf0L132

TerryE, 2015-12-22:
Nope.  It's one line of Lua.  Not going to implement.  If someone proposes a PR then we might consider it.

closed 2015-12-22 by TerryE

dvv, 2016-01-12:
Still +1 for toggle.
The rationale of toggle is i) _atomicity_ and ii) to down the toggle "execution" time (allows for faster bitbanging). The latter of course can be dirty worked around with a variable kept in sync with pin state.