nodemcu/nodemcu-firmware #547: getmac() new format

eyaleb, 2015-07-12:
Today I build from github and things were failing. Turns out the format of `wifi.sta.getmac()` changed from something like (this is what the wiki says)
    `mac address in string, for example:"18-33-44-FE-55-BB"`
to lower-case and a different separator
    `18:33:44:fe:55:bb`

BTW, `.setmac()` says upper-case with a colon separator, different again.

Was this intentional? It makes any app that uses the MAC to set parameters  (as mine does) not work.

Also, to be sure that I am using the correct SDK, where is the SDK zip file that master is built against?
`user_version.h` says "0.9.6" but I do not see a proper 0.9.6 package, the latest 0.9 series was 0.9.5 (with a later patch1).

TIA

jiegec, 2015-07-12:
This format is defined by MACSTR in user_interface.h which is "%02x:%02x:%02x:%02x:%02x:%02x" now. Maybe the wiki is outdated?

eyaleb, 2015-07-12:
While the code and wiki are out of sync, the gist of my issue is that the format changed since the earlier (March) release. Is this a good idea?
I now do this to work around the change
`mac = string.gsub(string.upper(wifi.sta.getmac()),":","-")`

marcelstoer, 2016-06-29:
Found this just now. I've no intention to rewrite history but I fixed the docs: https://nodemcu.readthedocs.io/en/latest/en/modules/wifi/#wifistagetmac. `wifi.sta.setmac()` is case-insensitive btw.

Usually one doesn't care whether a hex digit is lower or upper case. Separating the groups by `:` is just one of three more or less common formats for MAC addresses.

closed 2016-06-29 by marcelstoer