nodemcu/nodemcu-firmware #549: esp-07 does not work with 20150627 version

gbox3d, 2015-07-13:
esp-07 don't work with 
but work well esp-12

MarsTechHAN, 2015-07-13:
ESP07 use same chip and flash as ESP12, so there must be something wrong with yourself.(´･ᆺ･`)

gsalvati, 2015-07-13:
Maybe your ESP07 is the faulty one.. wrong position soldered oscilator

MarsTechHAN, 2015-07-13:
That's impossible... All the modules have been checked before shipping out.

vudangngoc, 2015-07-13:
Try to re-flash several times, maybe there are some thing wrong while flashing

gbox3d, 2015-07-14:
is there anyone esp-07? 
i try five of esp-07 ...... ,but all of esp-07 do not work.
it worked well old version firm(20150127)

Aeprox, 2015-07-14:
Running fine here on 2 different esp-07's.
Could you describe the problem in a bit more detail? "Not working" could be anything.

eyaleb, 2015-07-15:
@gbox3d
>esp-07 don't work

Come on! surely you can be more specific if you want help. Why should we labour your riddle?

Can you flash the fw? Can you then run any lua commands in a terminal before uploading your app? Can you upload your program? Does it work? If it fails, how is it failing?

All my esp-07 work just fine, but I select only the necessary modules in `app/include/user_modules.h` which makes the fw smaller in the first place.

@MarsTechHAN
>ESP07 use same chip and flash as ESP12

I disagree. While I did not try this fw (I build my own), all my -07 have 512KB and my -12 have 4MB. Could this be a flash size issue?

cheers

vudangngoc, 2015-10-26:
I also have problem with ESP7. I checkout dev140 branch and build it with esp-open-sdk ver 1.4.0. The image run well with ESP12, but cannot communicate with ESP7. I don't know why :(

marcelstoer, 2015-10-26:
I have heard that SDK 1.4 based builds (currently `dev` and `dev140`) don't fit onto 512KB boards but I can't confirm myself.

vudangngoc, 2015-10-26:
I removed most of module and 0x10000.bin has just 396803 bytes. I think it fit with 512KB. 
But it will very bad news for me if nodemcu will not support esp7

TerryE, 2015-10-26:
No, the 1.4 build will fit as long as you leave out SSL, and all of the modules that you don't need for that H/W config.  IOt works fine with my ESP-01s, so it should work OK with the 07s.

vudangngoc, 2015-10-27:
I use some modules, I think those modules don't require SSL. But still not work with ESP07

#define LUA_USE_MODULES_NODE
#define LUA_USE_MODULES_FILE
#define LUA_USE_MODULES_GPIO
#define LUA_USE_MODULES_WIFI
#define LUA_USE_MODULES_NET
#define LUA_USE_MODULES_TMR
#define LUA_USE_MODULES_ADC
#define LUA_USE_MODULES_CJSON

Aeprox, 2015-10-27:
No issues here with ESP07 when I disable SSL and some unneeded modules.

One pointer though; If you've already flashed a build that didnt fit into
flash (very likely if you've got a 4Mbit/512MByte module), you'll have to
reflash esp_init_data_default.bin, because flashing the "oversized" build
overwrote that section of flash memory.

vudangngoc, 2015-10-28:
I tried to flash dev140 on 2 other "fresh" ESP-07, but still not working. I should move on ESP-12 although I prefer ext anten of ESP-07.
Thank you for your support.

eyaleb, 2015-10-30:
I too have an issue here.

Now that dev branch in on 1.4.0 I am testing too. I flashed my esp-12 fine and it seems to work (but more testing is required). However using esp-07 does not boot.

What I get (on resent signal) is
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

ets_main.c
```
The flashing (preceded by a 512KB blanking):
```
esptool.py --port COM7 -b 115200 write_flash  0x07c000 esp_init_data_default.bin 0x00000 0x00000.bin 0x10000 0x10000.bin
Connecting...
Erasing flash...
Writing at 0x0007c000... (100 %)
Erasing flash...
Writing at 0x00007c00... (100 %)
Erasing flash...
Writing at 0x00075c00... (100 %)

Leaving...
```
As is seen above, the main code did not overwrite the parameters area. The files sizes are
```
  32,608 0x00000.bin
 417,363 0x10000.bin
  25,564 esptool.py
     128 esp_init_data_default.bin
```
What I changed: adjusted `user_config.h` to say:
```
#undef CLIENT_SSL_ENABLE
```
and removed many unused modules in `user_modules.h`.

Should I have done anything else?

TIA

vudangngoc, 2015-10-30:
@eyaleb : you should add #define FLASH_512K and remove #define DEVELOP_VERSION at user_config.h . You also need to reflash esp_init_data_default.bin at 0x7C000 because the "oversized" build overwrote that section of flash memory
https://github.com/espressif/ESP8266_AT/tree/master/bin

eyaleb, 2015-10-30:
Thanks @vudangngoc.

The default has
```lua
// #define DEVELOP_VERSION
```
and you can see that I had the `init_data` flashed too.

I made some progress though. Using `FLASH_512K` instead of `FLASH_AUTOSIZE` allowed me to test a small flash setup on an esp-12 (which actually has 4MB). It still did not boot on the esp-07. Removing a few more items from the `config` made it boot but had only 5KB flash for programs, I need about 25KB.

Removing the UART from `user_modules.h` turned out to be the big win, releasing over 60KB. I can now boot the esp-07 and run my application, though it is running much slower than expected, cycling in 2.45s compared to 0.72s with 0.9.5. This needs further investigation.

Interestingly, running the same on the esp-12 (so pretending it has only 512KB) goes very fast, cycling at 0.54s compared to 4.12s with 0.9.5 (and AUTOSIZE). More to look into.

vudangngoc, 2015-10-30:
My esp7 also get restart many times and have some log at 115200:

wdt reset
load 0x40100000, len 29672, room 16 
tail 8
chksum 0xed
load 0x3ffe8000, len 2264, room 0 
tail 8
chksum 0xbd
load 0x3ffe88d8, len 8, room 0 
tail 8
chksum 0xb8
csum 0xb8
rl��rl�
 ets Jan  8 2013,rst cause:4, boot mode:(3,4)

eyaleb, 2015-11-01:
@vudangngoc, wdt resets often means a program error. Maybe call tmr.wdclr() before doing lengthy work?

Regarding my earlier comment:

The speed issues are now resolved, was related to first time restart after a reset.
So far the best setup is an esp-12e (waiting for my 12f) with a very low deep sleep of 12uA, an MCP1700 LDO with 1.5uA Iq, and a firmware built (dev branch) with FLASH-512K. This leaves about 80KB of flash for storage. I upload *.lc files from lua.cross build to save space.

esp-07 works fine with this setup but has a higher Iq (about 30uA).

TerryE, 2015-12-12:
See #719

closed 2015-12-12 by TerryE