nodemcu/nodemcu-firmware #552: What is the status of dev120 branch?

eyaleb, 2015-07-14:
I see the two new `dev1*` branches. I do not see any description of their status (the README is still from master).

Is this work-in-progress? Is it ready for the riff-raff (e.g. myself) to try? Should I leave it alone while it takes shape?

cheers

marcelstoer, 2015-07-14:
Curious as well... Both CI builds are broken, though: #532.

gbox3d, 2015-07-14:
it may be sdk1.2 version?! right?

marcelstoer, 2015-07-14:
The SDK is defined by what is in https://github.com/nodemcu/nodemcu-firmware/blob/dev120/tools/esp-open-sdk.tar.gz. I can see it's still the same as in the master branch. So, I guess dev120 is meant to be working _towards_ SDK 1.2 but they're not there yet.

eyaleb, 2015-07-14:
Sure, I expected this to be the SDK v1.2 port. Still, what is the status? Can any of the contributors please comment or update the README?

So far I built it OK. Flashing was OK. Uploading files is shaky. My app needs float... rebuilt and `section '.text' will not fit in region iiram1_0_seg'`. Removed some (unused) modules and got it to build. I could then run my app. I had to break one large LUA file in two (it did not compile whole).

Running the app looks OK, but every so often if crashes with 'wrong arg range'. Don't know where but in general memory is more tight than master. The app is also noticeably slower (4.5s vs 3.5s).

I will leave it alone for a while hoping it will stabilise.

cheers

eyaleb, 2015-07-14:
I see the README now says `version 1.2.0 Under developing, Only integer firmware works.` Thanks.

georgkreimer, 2015-07-24:
It builds but doesn't really run on the ESP-01, ESP-12, ESP-12E and other that I have here

eyaleb, 2015-07-26:
To ensure any comments I make are relevant, can we please have the recommended build procedure, from cloning the repository to having the fw image. I assume that the repository already contains all the dependencies (including the SDK).

TIA

marcelstoer, 2015-07-26:
@eyaleb I don't think your comment is relevant for this issue in particular. It sounds more like a general question that should be asked elsewhere e.g. www.esp8266.com/viewforum.php?f=17 or http://stackoverflow.com/questions/tagged/nodemcu. At some point there should be instructions on how to build the firmware yourself on http://www.esp8266.com/wiki/doku.php?id=nodemcu-unofficial-faq I guess even though there are numerous blog posts by other people around the Internet.

eyaleb, 2015-07-26:
My question is not about generally building the fw (I do this regularly) but what this specific branch requires. Doing a straight 'make' is missing some headers and later some libs too. I point at the SDK directory which resolves everything. However, the SDK to use must be compatible with the ones as present in some files in the nodemcu /lib. Some instructions suggests copying files across but I prefer not to, this is what `-I` and `-L` is for.

So my question should probably say: what is the environment in which this fw should be built?

TerryE, 2015-10-31:
I will be deleting dev120 tomorrow unless any of the committers object.

closed 2015-10-31 by TerryE