nodemcu/nodemcu-firmware #554: Mqtt continuously sends PUBACK msg

jasonhector, 2015-07-14:
I am running nodemcu 0.96 integer custom build using http://frightanic.com/nodemcu-custom-build/.  I use RSMB MQTT broker and mqtt-spy client. 

I have stripped down the mqtt code to bare minimum and am able to reproduce the problem. 
When nodemcu received a publish with QOS 1 it sends a PUBACK [as it should] but then continues to send this PUBACK infinitely with a period of 1 sec - which it shouldnt.

My sample code is:
```lua
 -- mote.lua

clientId='esp/200'
mqttConnected =0            -- 0=not connected ; 1 = only connected ; 2= connected and subscribed
m = mqtt.Client(clientId,120, '','')--user,password)        --keepalive = 120

local function subscribed(con)
    mqttConnected =2
    m:publish(clientId.."/online", "1", 1, 1,function(conn) end)        --report device online
end

local function connected(con)
    mqttConnected =1
    m:subscribe(clientId..'/$L/+',1,subscribed)
end

m:lwt(clientId.."/online", "0", 1, 0)   --last will testatement
m:on("offline", function(con)  mqttConnected = 0  end)  
m:on("message", function(con, topic, data) end)

m:connect('192.168.2.11',1884, 0,connected)
```

jasonhector, 2015-08-01:
Will this be addressed by the developers?

karrots, 2015-10-12:
This issue is not the same as issue #561. I am also seeing this.

karrots, 2015-11-03:
This can be closed.

closed 2015-11-03 by TerryE