nodemcu/nodemcu-firmware #555: Not enough Memory issue.

roccomuso, 2015-07-14:
Hi guys,
can someone help me to optimize this code? I'm getting the "Not enough Memory" error during the node.compile('server.lua') command.

<code>

</code>

I have the latest NodeMCU firmware.

npadhya, 2015-07-14:
You can take a look at this file : https://github.com/npadhya/Home_Connect/blob/master/ESP_Web_Service.lua

I guess it's doing the same thing you are trying, and working fine for me.

NOTE: THIS IS NOT MY WORK!!. I Don't remember the site I have copied this code from. :)

Aeprox, 2015-07-14:
Your code seems to be appending a lot of strings, that's probably the cause. I suggest you read http://www.lua.org/pil/11.6.html , which describes a more efficient way of concatenating strings. It boils down to adding all your substrings to a table, and converting that table of strings into one big string in one go once you've added al substrings.

roccomuso, 2015-07-14:
@npadhya i know that code. But mine it's more complex.

jasonhector, 2015-07-14:
Give this a read as well to optomise your memory and heap usage:
http://www.esp8266.com/wiki/doku.php?id=nodemcu-unofficial-faq

roccomuso, 2015-07-14:
Does ESP-09 have more RAM?

TerryE, 2015-07-15:
@roccomuso, you can recode around the memory limitations as I describe in the unofficial FAQ.  This isn't a bug in the firmware per se, so this isn't an issue for this list.

MarsTechHAN, 2015-07-15:
All the ESP moudles use same chip; its full name is ESP8266EX, and have same SRAM. You need to try to break you program into serval modules, and use a main Lua program to control them. That will be better idea.

elefa, 2015-07-20:
I think too many string variables (method, path, vars). Try to work only with "request" and as few and as short string variables as possible.
This code get data from POST response (HTML form): 
    srv=net.createServer(net.TCP) 
    srv:listen(80,function(conn)
    conn:on("receive",function(conn,payload)
    payload=payload:sub(payload:find('\r\n\r\n')+4,#payload-2):gsub('\r\n','&')
    print(payload)

TerryE, 2015-07-20:
@elefa, this advice is a bit random and controversial.  Please stick to issues and comment relating to the development of the nodeMCU firmware itself on this issues list.

closed 2015-08-21 by TerryE